APEbbers/FreeCAD-Ribbon #45: There is python2 code in the addon. Are there plans of cleaning it up since py3 is the current standard?

luzpaz, 2024-10-24:
Just asking.

APEbbers, 2024-10-24:
Hi,

Yes. It should all be python3 code. Where did you find it? in the keyboard perhaps? if so, it might be better to use a different package.

APEbbers, 2024-10-24:
I found the tool [](https://docs.python.org/2/library/2to3.html). i did run it. 
It changed some imports to relative imports, which resulted in an error when loading freecad. So the imports remain absolute but for the rest, i've updated the code to Python3.

I skipped the local packages, because these are the packages from PyPi which I don't want to change too much.
They are local due to install issues with the addon-manager. Both "Keyboard" and "PyQTRibbon" supports python3, so that shouldn't be an issue.

closed 2024-10-24 by APEbbers