APEbbers/FreeCAD-Ribbon #56: Optionally add the ability to support searching using 3rd party SearchBar

luzpaz, 2024-10-27:
Using the existing https://github.com/SuzanneSoy/SearchBar could Ribbon UI have an optional dependency to leverage SearchBar functionality within it ?

APEbbers, 2024-10-28:
I look in to that!

luzpaz, 2024-10-28:
If this does become a thing you could add SearchBar  as an optional dependency into the [package.xml](https://wiki.freecad.org/Package_Metadata#.3Cdepend.3E) (Note: the syntax below is untested.)
```xml
<!-- Optional install of 3rd-party SearchBar to streamline ability to Search and find commands -->
<depend type="addon" optional="true">SearchBar Workbench</depend>`
```

APEbbers, 2024-11-08:
Hi @luzpaz,

I've implemented the searchbar into the ribbon (Development branch):
[image]

I've also made a PR to the Searchbar to make it compatible with QT6. Currently QtCore.QTextCodec is used which is no longer standard with QT6. (https://github.com/SuzanneSoy/SearchBar/pull/36).
Are you able to review and merge this PR?

luzpaz, 2024-11-08:
I'm not sure @suzannesoy is active in FreeCAD anymore. Perhaps you could fork it and take over maintainership ?

luzpaz, 2024-11-08:
@obelisk79 would you have a moment to test this new feature of RibbonUI ?

APEbbers, 2024-11-08:
> I'm not sure @SuzanneSoy is active in FreeCAD anymore. Perhaps you could fork it and take over maintainership ?
> 
> I'm also having issues with FreeCAD appimage and don't have Snap or flatpak ATM. Waiting for the appimage bug to get fixed

Sure I can over the maintainership, but I'll keep my focus to the FreeCAD Ribbon. I only do urgents bugfixes for the time being.
is that allright?

luzpaz, 2024-11-08:
@chennes SearchBar has a new fork/maintainer. lets point the addon manager toward it?

obelisk79, 2024-11-08:
> @obelisk79 would you have a moment to test this new feature of RibbonUI ?

probably not until some time next week. testing out this addon has been on my radar for several weeks anyway.

luzpaz, 2024-11-10:
cc @hasecilu

hasecilu, 2024-11-11:
I made a quick test on the new repo  but creashed FreeCAD when clicking on the search bar. Along with Ribbon from develop branch

```
OS: Arch Linux (Hyprland/hyprland/wayland)
Architecture: x86_64
Version: 1.1.0dev.36460 +2692 (Git)
Build type: Release
Branch: test/prs
Hash: f89281e3ff7a4aa75aa5b511b49f4b3058934b8c
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * FreeCAD-Ribbon 1.4.0
  * SearchBar 1.1.0
```

```gdb
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
Downloading 27.27 K source file /usr/src/debug/glibc/glibc/string/../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S
__memcpy_avx_unaligned () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:220
220		VMOVU	(%rsi), %VMM(0)
(gdb) bt
#0  __memcpy_avx_unaligned () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:220
#1  0x00007fffef0f4e7d in memcpy (__dest=0x7fffffff62b0, __src=<optimized out>, __len=3152, __dest=<optimized out>, __src=<optimized out>, __len=<optimized out>)
    at /usr/include/bits/string_fortified.h:29
#2  QBuffer::readData (this=<optimized out>, data=0x7fffffff62b0 '\376' <repeats 200 times>..., len=3152) at /usr/src/debug/qt6-base/qtbase/src/corelib/io/qbuffer.cpp:389
#3  0x00007fffef121736 in QIODevicePrivate::read (this=0x55555aff7290, data=0x7fffffff62b0 '\376' <repeats 200 times>..., maxSize=<optimized out>, peeking=false)
    at /usr/src/debug/qt6-base/qtbase/src/corelib/io/qiodevice.cpp:1076
#4  0x00007fffef1f6c68 in QTextStreamPrivate::fillReadBuffer (this=0x55555e897450, maxBytes=-1) at /usr/src/debug/qt6-base/qtbase/src/corelib/serialization/qtextstream.cpp:353
#5  0x00007fffef1ffe67 in QTextStreamPrivate::read (this=0x55555e897450, maxlen=2147483647) at /usr/src/debug/qt6-base/qtbase/src/corelib/serialization/qtextstream.cpp:507
#6  QTextStream::readAll (this=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/serialization/qtextstream.cpp:1516
#7  0x00007fffbc4fb179 in ??? () at /usr/lib/python3.12/site-packages/PySide6/QtCore.cpython-312-x86_64-linux-gnu.so
#8  0x00007ffff15859db in cfunction_vectorcall_NOARGS (func=0x7fff53f85ee0, args=<optimized out>, nargsf=<optimized out>, kwnames=0x0) at ./Include/cpython/methodobject.h:50
#9  0x00007ffff15a68bd in _PyObject_VectorcallTstate
    (tstate=0x7ffff1a2cdb0 <_PyRuntime+458992>, callable=0x7fff53f85ee0, args=0x7ffff7f90918, nargsf=9223372036854775808, kwnames=0x0) at ./Include/internal/pycore_call.h:92
#10 PyObject_Vectorcall (callable=0x7fff53f85ee0, args=0x7ffff7f90918, nargsf=9223372036854775808, kwnames=0x0) at Objects/call.c:325
#11 0x00007ffff158b5b7 in _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at Python/bytecodes.c:2715
#12 0x00007ffff15e38ec in _PyEval_EvalFrame (tstate=0x7ffff1a2cdb0 <_PyRuntime+458992>, frame=<optimized out>, throwflag=0) at ./Include/internal/pycore_ceval.h:89
#13 _PyEval_Vector (tstate=0x7ffff1a2cdb0 <_PyRuntime+458992>, func=0x7fff5d307560, locals=0x0, args=0x7fffffffa630, argcount=2, kwnames=0x0) at Python/ceval.c:1683
#14 _PyFunction_Vectorcall (func=0x7fff5d307560, stack=0x7fffffffa630, nargsf=2, kwnames=0x0) at Objects/call.c:419
#15 _PyObject_VectorcallTstate (tstate=0x7ffff1a2cdb0 <_PyRuntime+458992>, callable=0x7fff5d307560, args=0x7fffffffa630, nargsf=2, kwnames=0x0)
    at ./Include/internal/pycore_call.h:92
#16 method_vectorcall (method=<optimized out>, args=<optimized out>, nargsf=<optimized out>, kwnames=0x0) at Objects/classobject.c:91
#17 0x00007ffff15905ea in PyCFunction_Call (callable=0x7fff8016c580, args=0x7fff80137460, kwargs=0x7fff681bba00) at Objects/call.c:387
#18 _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at Python/bytecodes.c:3263
#19 0x00007ffff15e38ec in _PyEval_EvalFrame (tstate=0x7ffff1a2cdb0 <_PyRuntime+458992>, frame=<optimized out>, throwflag=0) at ./Include/internal/pycore_ceval.h:89
#20 _PyEval_Vector (tstate=0x7ffff1a2cdb0 <_PyRuntime+458992>, func=0x7fff80af9800, locals=0x0, args=0x7fffffffa8a0, argcount=2, kwnames=0x0) at Python/ceval.c:1683
#21 _PyFunction_Vectorcall (func=0x7fff80af9800, stack=0x7fffffffa8a0, nargsf=2, kwnames=0x0) at Objects/call.c:419
#22 _PyObject_VectorcallTstate (tstate=0x7ffff1a2cdb0 <_PyRuntime+458992>, callable=0x7fff80af9800, args=0x7fffffffa8a0, nargsf=2, kwnames=0x0)
    at ./Include/internal/pycore_call.h:92
#23 method_vectorcall (method=<optimized out>, args=<optimized out>, nargsf=<optimized out>, kwnames=0x0) at Objects/classobject.c:91
#24 0x00007ffff15a68bd in _PyObject_VectorcallTstate (tstate=0x7ffff1a2cdb0 <_PyRuntime+458992>, callable=0x7fff80cda980, args=0x7fffffffa960, nargsf=1, kwnames=0x0)
    at ./Include/internal/pycore_call.h:92
#25 PyObject_Vectorcall (callable=0x7fff80cda980, args=0x7fffffffa960, nargsf=1, kwnames=0x0) at Objects/call.c:325
#26 0x00007fffb0b87d42 in ??? () at /usr/lib/python3.12/site-packages/PySide6/QtWidgets.cpython-312-x86_64-linux-gnu.so
#27 0x00007ffff01510c9 in QWidget::event (this=0x55555d12e1f0, event=0x7fffffffade0) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidget.cpp:9086
#28 0x00007ffff00fe31a in QApplicationPrivate::notify_helper (this=<optimized out>, receiver=0x55555d12e1f0, e=0x7fffffffade0)
    at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3294
[... cut ...]
#57 0x00007ffff4df8bed in (anonymous namespace)::tryRunEventLoop (mainApp=...) at /home/uli/git/FreeCAD/FreeCAD/src/Gui/Application.cpp:2196
#58 0x00007ffff4df8f47 in (anonymous namespace)::runEventLoop (mainApp=...) at /home/uli/git/FreeCAD/FreeCAD/src/Gui/Application.cpp:2225
#59 0x00007ffff4df9300 in Gui::Application::runApplication () at /home/uli/git/FreeCAD/FreeCAD/src/Gui/Application.cpp:2302
#60 0x00005555555795eb in main (argc=1, argv=0x7fffffffcdc8) at /home/uli/git/FreeCAD/FreeCAD/src/Main/MainGui.cpp:330
```

APEbbers, 2024-11-11:
Bit strange, i just tested it on a linux debian with appimage version 39153 and it works as expected.
I know that on windows, the searchbar is currently not compatible with QT6. I have a fork with a fix for that:[https://github.com/APEbbers/SearchBar.git](url)

Maybe you can try this to see if it stil happens. or perhaps you can share me your json file? I can then test if it also happens on my side?

hasecilu, 2024-11-11:
Hi, the previous test was made on self-compiled version on latest version `HEAD` with Qt6, now just tested again on appimage 39153 and in fact, it does work pretty well. So it could be there are only problems with Qt6 versions.

In any case, this is my file: 
[RibbonStructure.json](https://github.com/user-attachments/files/17708360/RibbonStructure.json)

luzpaz, 2024-11-13:
https://github.com/APEbbers/SearchBar.git is the correct url

closed 2024-11-20 by APEbbers