APEbbers/FreeCAD-Ribbon #69: Trying to install FreeCAD-Ribon with error message

ucool321, 2024-10-31:
Hello

I've installed FreeCAD-Ribon with the Addon Manager. But at Start i will get the error message "QWidget::setMinimumSize: (Ribbon/QDockWidget) Negative sizes (0,-1) are not possible". 

There are no toolbars selected.  If i active the Panel "Ribbon". I can't select/add any toolbar anymore.

APEbbers, 2024-10-31:
Hi,

Which OS are you using? And can you perhaps share a screenshot of the error message? 
You can try my develop branch in the meantime. I'm not loading the ribbon as a QDockwidget in that version.

ucool321, 2024-10-31:
Here is a screenshot.

[image]

I using it on Linux.

**Version:** Freecad 1.0rc2
**System spec:**
CPU: quad core Intel Core i7-6700K (-MT MCP-) speed/min/max: 982/800/4200 MHz
Kernel: 6.6.58-1-MANJARO x86_64 
Mem: 32 GiB (25.3%)

APEbbers, 2024-10-31:
#54

closed 2024-10-31 by APEbbers

ucool321, 2024-10-31:
I've installed the new version.

For my understanding, the toolbar icons will be automaticly added in the "Ribbon" Panel? I made a screenshot and the ribbon panel stays empty. The "QWidget" error message is gone. See the second screenshot.

[image]

[image]

APEbbers, 2024-10-31:
I'm not sure what is happening. do you have toggled overly by any change? You can check this by pressing "F3".
If so, the ribbon will be somewhere in the topleft corner. 
There is an issue with overlays which I'm working on.

ucool321, 2024-10-31:
Yes I thing this has something to do with issue [54](https://github.com/APEbbers/FreeCAD-Ribbon/issues/54) what you posted earlier. May be its a upstream bug in Freecad.