APEbbers/FreeCAD-Ribbon #77: Opening "Ribbon design" reloads all WB again when its second time

hasecilu, 2024-11-01:
The first time you open that dialog you are asked to load all WB, if you close it and then open it again you're asked again to load all WB when they were already loaded the first time, as consequence you have blocked the UI for another minute (in my case), **is it really needed to load all again?**

[image]

APEbbers, 2024-11-15:
Look to serialize function in SearchBar wb

hasecilu, 2024-11-17:
Do you mentioned because of the `iconToHTML()` function? I made few size changes (128px) and notice that on icon list the icons is well rendered but on other cases is slightly pixelated.

[image]
[image]

Well I noted that tooltips on SearchBar appear translated, thing I didn't achieve on FreeGrid WB, was that?

I'll check it more in detail to make those fixes there, thanks!

APEbbers, 2024-11-17:
I made actually a note to myself.  I believe the serialize function from the SearchBar addon can reduce the loading time of the design menu.

But regarding the translated tooltips. I found out that when you take the QAction from a command, and then QAction.text(), you get the translated text instead of english. 

To be fair, I didn't do anything to the searchbar except a QT6 fix in the serialize code. That fix was already there.

hasecilu, 2024-11-27:
Tried testing but both `Develop` and `Develop--Serialize` crash . Not sure if it's related with the dat file.

hasecilu, 2024-11-28:
It just closes.
### Serialize branch
No icons on some commands (fixxed on main develop)

[image]

### SIGSEGV on Develop
```gdb
(gdb) bt
#0  __memcpy_avx_unaligned () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:220
#1  0x00007ffff2cf4e7d in memcpy (__dest=0x7fffffff4760, __src=<optimized out>, __len=1920, __dest=<optimized out>, __src=<optimized out>, __len=<optimized out>)
    at /usr/include/bits/string_fortified.h:29
#2  QBuffer::readData (this=<optimized out>, data=0x7fffffff4760 '\376' <repeats 200 times>..., len=1920) at /usr/src/debug/qt6-base/qtbase/src/corelib/io/qbuffer.cpp:389
#3  0x00007ffff2d21736 in QIODevicePrivate::read (this=0x555571f0ee00, data=0x7fffffff4760 '\376' <repeats 200 times>..., maxSize=<optimized out>, peeking=false)
    at /usr/src/debug/qt6-base/qtbase/src/corelib/io/qiodevice.cpp:1076
#4  0x00007ffff2df6c68 in QTextStreamPrivate::fillReadBuffer (this=0x555573bc4f20, maxBytes=-1) at /usr/src/debug/qt6-base/qtbase/src/corelib/serialization/qtextstream.cpp:353
#5  0x00007ffff2dffe67 in QTextStreamPrivate::read (this=0x555573bc4f20, maxlen=2147483647) at /usr/src/debug/qt6-base/qtbase/src/corelib/serialization/qtextstream.cpp:507
#6  QTextStream::readAll (this=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/serialization/qtextstream.cpp:1516
#7  0x00007fffc04fb179 in ??? () at /usr/lib/python3.12/site-packages/PySide6/QtCore.cpython-312-x86_64-linux-gnu.so
#8  0x00007ffff45859db in cfunction_vectorcall_NOARGS (func=0x7fff034b5d50, args=<optimized out>, nargsf=<optimized out>, kwnames=0x0) at ./Include/cpython/methodobject.h:50
#9  0x00007ffff45a68bd in _PyObject_VectorcallTstate
    (tstate=0x7ffff4a2cdb0 <_PyRuntime+458992>, callable=0x7fff034b5d50, args=0x7ffff7f925c8, nargsf=9223372036854775808, kwnames=0x0) at ./Include/internal/pycore_call.h:92
#10 PyObject_Vectorcall (callable=0x7fff034b5d50, args=0x7ffff7f925c8, nargsf=9223372036854775808, kwnames=0x0) at Objects/call.c:325
#11 0x00007ffff458b5b7 in _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at Python/bytecodes.c:2715
#12 0x00007ffff4585706 in _PyEval_EvalFrame (tstate=0x7ffff4a2cdb0 <_PyRuntime+458992>, frame=0x7ffff7f920e0, throwflag=0) at ./Include/internal/pycore_ceval.h:89
#13 _PyEval_Vector (tstate=0x7ffff4a2cdb0 <_PyRuntime+458992>, func=0x7fff85bfa160, locals=0x0, args=0x7fffffff8b50, argcount=<optimized out>, kwnames=0x0) at Python/ceval.c:1683
#14 _PyFunction_Vectorcall (func=0x7fff85bfa160, stack=0x7fffffff8b50, nargsf=<optimized out>, kwnames=0x0) at Objects/call.c:419
#15 _PyObject_FastCallDictTstate (tstate=<optimized out>, callable=0x7fff85bfa160, args=0x7fffffff8b50, nargsf=<optimized out>, kwargs=<optimized out>) at Objects/call.c:133
#16 0x00007ffff45c1c67 in _PyObject_Call_Prepend
    (tstate=0x7ffff4a2cdb0 <_PyRuntime+458992>, callable=0x7fff85bfa160, obj=0x7fff85af00e0, args=0x7ffff49cf410 <_PyRuntime+75600>, kwargs=0x0) at Objects/call.c:508
#17 slot_tp_init (self=0x7fff85af00e0, args=0x7ffff49cf410 <_PyRuntime+75600>, kwds=0x0) at Objects/typeobject.c:9026
#18 0x00007ffff4582818 in type_call (type=<optimized out>, args=0x7ffff49cf410 <_PyRuntime+75600>, kwds=0x0) at Objects/typeobject.c:1679
#19 _PyObject_MakeTpCall (tstate=0x7ffff4a2cdb0 <_PyRuntime+458992>, callable=0x5555598958c0, args=<optimized out>, nargs=<optimized out>, keywords=<optimized out>)
    at Objects/call.c:240
#20 0x00007ffff458b5b7 in _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at Python/bytecodes.c:2715
#21 0x00007ffff45e3e85 in _PyEval_EvalFrame (tstate=0x7ffff4a2cdb0 <_PyRuntime+458992>, frame=0x7ffff7f92020, throwflag=0) at ./Include/internal/pycore_ceval.h:89
#22 _PyEval_Vector (tstate=0x7ffff4a2cdb0 <_PyRuntime+458992>, func=0x7fff85ba2de0, locals=0x0, args=0x7fffffff8e88, argcount=1, kwnames=<optimized out>) at Python/ceval.c:1683
#23 _PyFunction_Vectorcall (func=0x7fff85ba2de0, stack=0x7fffffff8e88, nargsf=<optimized out>, kwnames=<optimized out>) at Objects/call.c:419
#24 _PyObject_VectorcallTstate (tstate=<optimized out>, callable=0x7fff85ba2de0, args=0x7fffffff8e88, nargsf=<optimized out>, kwnames=<optimized out>)
    at ./Include/internal/pycore_call.h:92
#25 0x00007ffff45e3988 in method_vectorcall (method=<optimized out>, args=0x7ffff49cf428 <_PyRuntime+75624>, nargsf=0, kwnames=0x0) at Objects/classobject.c:69
#26 0x00007ffff57ccda8 in ??? () at /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8
#27 0x00007ffff57cf668 in PySide::SignalManager::callPythonMetaMethod(QMetaMethod, void**, _object*) () at /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8
#28 0x00007ffff57d5327 in PySide::SignalManager::qt_metacall(QObject*, QMetaObject::Call, int, void**) () at /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8
#29 0x00007ffff2db22b4 in doActivate<false> (sender=<optimized out>, signal_index=<optimized out>, argv=<optimized out>)
    at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobject.cpp:4146
#30 0x00007ffff3766524 in QAction::triggered (this=<optimized out>, _t1=<optimized out>) at /usr/src/debug/qt6-base/build/src/gui/Gui_autogen/include/moc_qaction.cpp:480
#31 QAction::activate (this=0x55555efa5d00, event=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/gui/kernel/qaction.cpp:1104
[... cut ...]
#93 0x00007ffff5daf3b0 in (anonymous namespace)::tryRunEventLoop (mainApp=...) at /usr/src/debug/freecad/freecad/src/Gui/Application.cpp:2196
#94 (anonymous namespace)::runEventLoop (mainApp=...) at /usr/src/debug/freecad/freecad/src/Gui/Application.cpp:2225
#95 Gui::Application::runApplication () at /usr/src/debug/freecad/freecad/src/Gui/Application.cpp:2302
#96 0x000055555555acd4 in main (argc=<optimized out>, argv=0x7fffffffcdc8) at /usr/src/debug/freecad/freecad/src/Main/MainGui.cpp:295
```

APEbbers, 2024-11-28:
Which version of FreeCAD are you using? "Std_MacroStopRecord" is from FreeCAD 0.21.
FreeCAD 1.0 doesn't have this command.

I have however added extra try-except statements. Maybe this helps?

hasecilu, 2024-11-28:
I'm on FreeCAD 1.0.0 and dev version, probably a external WB used (after removing all WB didn't appear again message which with try block was converted to warning). Crash still happens.

I put some `print()` and the problem seems to be here:

```python
        # --- Serialize Icons ------------------------------------------------------------------------------------------
        #
        WorkbenchIcon = []
        for WorkBenchItem in self.List_Workbenches:
            WorkBenchName = WorkBenchItem[0]
            Icon = Gui.getIcon(WorkBenchItem[1])
            if Icon is not None:
                SerializedIcon = Serialize.serializeIcon(Icon)

                WorkbenchIcon.append([WorkBenchName, SerializedIcon])
                # add the icons also to the deserialized list
                self.List_WorkBenchIcons.append([WorkBenchName, Icon])

```

APEbbers, 2024-11-29:
Hi,
Thanks! 
I'll compare the functions and update both functions in FreeCAD Ribbon and SearchBar.
I see that now only .png files are allowed. But be aware thatmost pixmaps are .svg.
This morning I've added try-catch statements to the section you mentioned, just to be sure.

You don't have any problems with the searchbar add-on? 

> bwt, the dot to make hidden files on the Backups dir and the RibbonDataFile.dat file is on purpose?

No, the .dat file is actually a json file, the json package adds a dot to it. On windows, you an still see the files. I'll change it, so that they are also visible on Linux.

hasecilu, 2024-11-30:
Seems the list of results is invisible and current selection is very far from search bar

[image]

APEbbers, 2024-11-30:
Hi

I noticed that because both FreeCAD Ribbon and SearchBar uses Serialize.py, this crashed FreeCAD on linux.
I've renamed the files so that they cannot be mixed.

Perhaps this helps also with your bug? I tested it on both Linux and Windows, and it seems fine on my side.
If it still is a problem, I'll make an issue for it at the SearchBar add-on

hasecilu, 2024-12-01:
That's a problem that happens when WB with the old style have modules with the same name. WB with the new structure introduced by looooo a while ago don't have this problem.

APEbbers, 2024-12-01:
Hi

just double checkking: Did you generate the cache the first time?
[image]

APEbbers, 2024-12-01:
Hi, Indeed it is still the old repo. I've made a ticket for it.!

hasecilu, 2024-12-01:
I already made a PR that I hope fixes it.

hasecilu, 2024-12-03:
Tested again and seems now WB ribbons are properly set.

APEbbers, 2024-12-03:
> Tested again and seems now WB ribbons are properly set.

I noticed that the strange ribbon layout was often that of the Test workbench. When I had the Test workbench disabled, this wrong loading occurred far less often.

Now I'm actually removing the Test Workbench during startup. I hope this will solve most of the loading issues

closed 2024-12-23 by APEbbers