APEbbers/FreeCAD-Ribbon #79: FR: Create specification for WB maintainers to have a json for their WB ribbon

hasecilu, 2024-11-01:
I'm currently maintaining [FreeGrid](https://github.com/instancezero/in3dca-freegrid "Freegrid") and [STEMEFIE](https://github.com/bilbaomakers/StemfieWB "STEMFIE") WBs and I'm playing with the customization for those WB. I would like that users that are using **Ribbon** when switching to one of them will be using some of those presets, I thinks a PR with the changes is enough but I think it's better that WB maintainers could include a JSON file on the root of their WB, Ribbon then will read that file and apply that preset. The second option is better because when the WB maintainer makes changes on the WB it could update also that JSON file without need to send another PR here.

**WB** preset should have priority over **Ribbon** preset and **user** presets should have priority over **WB** and **Ribbon**.

[image]
[image]

Question: how to remove "individual views"?

asda

APEbbers, 2024-11-01:
Hi,

There is a stylesheet set, which you can change under Ribbon preferences. If you select "base.qss" instead of "default.qss" does that change anything?
[image]

regarding "individual views", in FC 1.0 you can excluded under exclude panels:
[image]

in FC0.21 this toolbar doesn't exists so I created one. But i'm not sure if you can remove it like above.

hasecilu, 2024-11-01:
> There is a stylesheet set, which you can change under Ribbon preferences. If you select "base.qss" instead of "default.qss" does that change anything?

The usage of `base.qss` provoked this:

[image]

> regarding "individual views", in FC 1.0 you can excluded under exclude panels:

It worked!

hasecilu, 2024-11-01:
Sure, no hurry

APEbbers, 2024-11-09:
Hi,
I added the location of the json file to the parameters. You can retrieve it by:

> "preferences = App.ParamGet("User parameter:BaseApp/Preferences/Mod/FreeCAD-Ribbon")"
> "preferences.GetString("RibbonStructure")"

I've also set the order of tabs in the parameters:
> "preferences = App.ParamGet("User parameter:BaseApp/Preferences/Mod/FreeCAD-Ribbon")"
> "preferences.GetString("TabOrder")"

Currently it is in my develop branch only

hasecilu, 2024-11-11:
I just saw the change, looks good.

I was thinking on having a file with contents like this on my WB and that exact structure was copied/used to generate the ribbon for the WB for the first time in that user setup. Then if user changes it, cool.

I was thinking on an automatic search for this presets on each WB. ALternatively the default file that Ribbon WB proposes can be tweaked. For example. I think a better proposal for "Structure" panel would be to have all 4 icons "medium", this way the space is better used.

|before|after|
|-----|------|
|[image]| [image] |
```json
"FreeGridWorkbench": {
    "toolbars": {
        "order": [
            "FreeGrid"
        ],
        "FreeGrid": {
            "order": [
                "Storage box",
                "Bit cartridge holder",
                "Storage grid",
                "3_separator_FreeGridWorkbench",
                "Sketch",
                "Preferences page",
                "About FreeGrid"
            ],
            "commands": {
                "FreeGrid_StorageBox": {
                    "size": "large",
                    "icon": "box"
                },
                "FreeGrid_BitCartridgeHolder": {
                    "size": "large",
                    "icon": "holder"
                },
                "FreeGrid_StorageGrid": {
                    "size": "large",
                    "icon": "grid"
                },
                "FreeGrid_Sketch": {
                    "size": "large",
                    "icon": "sketch"
                },
                "FreeGrid_PreferencesPage": {
                    "size": "medium",
                    "icon": "preferences_page"
                },
                "FreeGrid_About": {
                    "size": "medium",
                    "icon": "about"
                }
            }
        }
    }
}
```

APEbbers, 2024-11-17:
Hi, 
in the latest update (develop branch) I've added two new scripts. With one you can order specific panels and with the other script you can set the icon size for commands in specific panels/toolbars. maybe you can use them in your WB?

The scripts are looking only in the json file. they won't add toolbars currently not present in the json file. But you can change them so that they do