APEbbers/FreeCAD-Ribbon #82: Bug: Modifying `RibbonStructure.json` makes the FreeCAD Addon Manager to backup Ribbon WB directory

hasecilu, 2024-11-01:
Addon Manager can not do a clean pull because of the changes and clones the repo in another directory, this change makes the changes from users "lost", they need to manually copy that file again.

I think all user changes should be in a separated file, even separated in a WB-basis to prevent this issue.

Maybe even considering to put config files on `$XDG_CONFIG_HOME/FreeCAD/` would be a good option.

[image]

APEbbers, 2024-11-01:
If added the .json files to gitignore. When "RibbonStructure.json" is not present, a copy and rename action will be done with "RibbonStructure.txt".

I believe this should solve the issue

closed 2024-11-01 by APEbbers

hasecilu, 2024-11-01:
Yeah, I also think that should work. I can test when there is another change on main.

hasecilu, 2024-11-01:
OK, it seems since the file is already commited, git tracks its changes, it should be removed from repo or use another name for the file generated by the user.

APEbbers, 2024-11-01:
I've removed the files from the repo, so all .json files shouldn't be tracked anymore