APEbbers/FreeCAD-Ribbon #93: Bug: Replace hardcoded icon paths with current paths

hasecilu, 2024-11-06:
When the custom STEMFIE Ribbon was created the used icons were hardcoded on the JSON file, as consequence now that I'm updating bad icons on the WB the new icons instead of appearing it does appear a `?` symbol.

 The other 3 icons that DO APPEAR is because those commands were renamed, but the ones that conserved the name still try to use the old hardcoded value.

Also I think would be better to use relative paths.

[image]

hasecilu, 2024-11-06:
Updated to latest version and it's the same. Even the ribbon tool tries to load old icons

[image]

APEbbers, 2024-11-06:
I'm looking for an solution, but i find it still strange. I'm not manipulating the icons. The design menu just reads the information from FreeCAD. Some workbenches present absolute paths and other relative. Below a picture on how i get the icon pixmaps:
[image]

I'll install your wb to do some tests.

hasecilu, 2024-11-06:
Yes, I noticed that, check my edited comment above, probably I'll change those icons to use relative path, is possible to overwrite the icons tag on the JSON file? If the icon is read from pixmap when ribbon is created could be possible to overwrite old data.

closed 2024-11-06 by APEbbers

APEbbers, 2024-11-06:
Yes that can indeed. In the design dialog, go to the ribbon design tab. Then select the workbench and toolbar. When you then press update, the tags will be overwritten in the JSON file. 

Another option is to manually remove the tags or the complete WB from the json file.

hasecilu, 2024-11-06:
Cool! the Update button did the trick.
[image]