APEbbers/FreeCAD-Ribbon #99: Ribbon Design can't load all workbenches with Fasteners workbench installed

SpeeQz1, 2024-11-06:
When trying to open the `Customize... -> Ribbon Design` menu with the Fasteners workbench installed on latest weekly build of FreeCAD there is an error and the program has to be killed to fix the issue otherwise it keeps repeating:

[image]

The Lattice 2 workbench on the other hand works fine.

FreeCAD-Bundle AppImage: weekly-build
Addons: OpenTheme ( + OpenTheme preferences pack ), Lattice 2 Workbench, Fasteners Workbench
OS: Linux Mint 22 ( Cinnamon )

APEbbers, 2024-11-06:
Which version of the Ribbon are you using? I  just found a bug and created an update. I believe it was that exact issue.
This is my latest version:
[image]

SpeeQz1, 2024-11-06:
My bad jumped the gun a bit too quickly on that.
Update from 1.3.16.1 to 1.3.17 and the issue still persists.

APEbbers, 2024-11-06:
Does the report view show a more detailed error? I have no problem on my side, but maybe in the report view, a line number is mentioned.

APEbbers, 2024-11-06:
I did add some extra exception handling. can you try version 1.3.17.1?

SpeeQz1, 2024-11-06:
```sh
user@user-MS-7E07:~$ '/home/user/AppImages/freecad.appimage' 
FreeCAD 1.1.0, Libs: 1.1.0devR39127 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Activating Ribbon Bar...
FreeCAD Ribbon: https://github.com/APEbbers/FreeCAD-Ribbon
'NoneType' object has no attribute 'children'
'NoneType' object has no attribute 'children'
'NoneType' object has no attribute 'children'
"Empty widget item in QGridLayout 'gridLayout_8'."
'NoneType' object has no attribute 'children'
Traceback (most recent call last):
  File "/home/user/.local/share/FreeCAD/Mod/fasteners/./InitGui.py", line 112, in Activated
    FastenerBase.InitCheckables()
  File "/home/user/.local/share/FreeCAD/Mod/fasteners/./FastenerBase.py", line 890, in InitCheckables
    matchOuterButton = FSGetToolbarItem(commandsToolbarText, matchOuterButtonText)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/.local/share/FreeCAD/Mod/fasteners/./FastenerBase.py", line 241, in FSGetToolbarItem
    for c in mw.children():
             ^^^^^^^^^^^

shutil.which('openscad') did not return a result. Using fallback.
OpenSCAD executable not found
'NoneType' object has no attribute 'children'
Traceback (most recent call last):
  File "/home/user/.local/share/FreeCAD/Mod/fasteners/./InitGui.py", line 112, in Activated
    FastenerBase.InitCheckables()
  File "/home/user/.local/share/FreeCAD/Mod/fasteners/./FastenerBase.py", line 890, in InitCheckables
    matchOuterButton = FSGetToolbarItem(commandsToolbarText, matchOuterButtonText)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/.local/share/FreeCAD/Mod/fasteners/./FastenerBase.py", line 241, in FSGetToolbarItem
    for c in mw.children():
             ^^^^^^^^^^^

'NoneType' object has no attribute 'children'
Traceback (most recent call last):
  File "/home/user/.local/share/FreeCAD/Mod/fasteners/./InitGui.py", line 112, in Activated
    FastenerBase.InitCheckables()
  File "/home/user/.local/share/FreeCAD/Mod/fasteners/./FastenerBase.py", line 890, in InitCheckables
    matchOuterButton = FSGetToolbarItem(commandsToolbarText, matchOuterButtonText)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/.local/share/FreeCAD/Mod/fasteners/./FastenerBase.py", line 241, in FSGetToolbarItem
    for c in mw.children():
             ^^^^^^^^^^^
```

APEbbers, 2024-11-06:
Hi,

That issue is not related to the FreeCAD Ribbon, but an issue in the Fasteners WB itself.  It is probably an issue with FreeCAD 1.0.
I've just tested FreeCAD 0.21 and then there is not an issue.

I also noticed that when you set the fasteners wb to be loaded at startup, that the issue is gone. So that is an easy fix!
Probally when FreeCAD 1.0 is ready for release the bug will be solved.

closed 2024-11-07 by APEbbers