APEbbers/FreeCAD-Ribbon #103: Ribbon preferences removing tabbar

SpeeQz1, 2024-11-09:
I remember this issue being quite old, and wanted to know are there any plans on implementing a system on somehow recovering preferences so that the tabbar would be properly returned when removing/disabling the Ribbon UI addon?

[image]
[image]

APEbbers, 2024-11-10:
Hi,

I'm aware of this issue. Personally i think it is a missing feature of FreeCAD itself. Within the FreeCAD Ribbon add-on I cannot do much, but perhaps a separate add-on can provide this functionality.  I can imagine that you always want to have a restore point, in case FreeCAD crashes and is meshed up as a result.

For the moment there is a script that i have mentioned in my wiki-page: [https://github.com/APEbbers/FreeCAD-Ribbon/wiki/01-%E2%80%90-Installation#uninstallation](url).

This also an old "solution" that i took over for the time being.

APEbbers, 2025-05-04:
Hi @SpeeQz1,

It has been a while but for this issue I've released a separate add-on: [https://github.com/APEbbers/SaveAndRestore](url).
With this add-on you can save and restore the config files which contains all settings. But most importantly, you can restore all toolbars after you disabled or uninstalled the Ribbon UI. 

With the next release of the Ribbon UI, the search and restore add-on, will be added as an optional dependency, like with the searchbar. You can the find the save and restore button under the Tools menu and when the Ribbon is installed, under the preference button in the top right corner.

The add-on is available in the add-on manager. I believe this solves the issue. 
I hope you like the add-on.

Kind regards

closed 2025-05-07 by APEbbers