APEbbers/FreeCAD-Ribbon #105: Changing available workbenches with Ribbon Design messes up their contents

SpeeQz1, 2024-11-11:
When I tried removing workbenches from the ribbon my Spreadsheet workbench got populated with other workbench buttons and they can't be removed.

[image]
[image]
[image]

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39141 (Git) AppImage
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.29 (Disabled)
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * CurvedShapes 1.0.13
  * FreeCAD-Ribbon 1.4.0
  * SearchBar 1.0.2
  * Backups
  * OpenTheme 2024.9.1
```

APEbbers, 2024-11-11:
Hi,

i've a difficulty reproducing this error. From my side it works on both Linux Debian as on Windows. Perhaps you can share your "RibbonStructure.json" file with me? it is located in the folder of the add-on. 

To fix it, you can als go back one version with the restore button. Every time you press update or close a backup is created.
[image]

APEbbers, 2024-11-11:
You probably have the backup function disabled. You can enable it under preferences. I make it so, that it will be enabled at first install for the future.

Your RibbonStructure.json looks fine to me. Can you remove or move it out the folder of the addon? With the develop version, a new file will be created. Or press the reset button in the design dialog that should replace the json file with a new one.

SpeeQz1, 2024-11-11:
> You probably have the backup function disabled. You can enable it under preferences. I make it so, that it will be enabled at first install for the future.
>
> Your RibbonStructure.json looks fine to [...]

Resetting the ribbon design fixes everything it's that for some specific reason only Spreadsheet seems to fail when adding/removing workbenches to the ribbon.

SpeeQz1, 2024-11-11:
I figured something out.
I reset the Workbenches preferences and the Spreadsheet workbench was fixed, in Ribbon. It seems there's some connection between which workbenches you have enabled/disabled.

[image]

APEbbers, 2024-11-11:
Aha, That's good to here.  it sounds like a mixup in FreeCAD internally. The ribbon retrieves the toolbars from the active workbench to build the panels from. It looks like as if FreeCAD returned the wrong set of toolbars. It can be a cache issue what you fixed by reset the workbenches.

I keep it in mind and check if I see something similar. For now I close this issue. If it occurs again it can be reopend

> Have you thought about joining the FreeCAD Discord, that should help with speeding up the debugging process.

I'm personally prefer the issues being raised here at github. This way i have more overview. I'm doing this in the evening hours, for me the speed is good enough at the moment☺!

Thanks for the help anyway!

closed 2024-11-11 by APEbbers