APEbbers/FreeCAD-Ribbon #110: Smal & Mediuml ribbon buttons disappearing

SpeeQz1, 2024-11-17:
For some reason specifically small ribbon buttons like "Add a Primitive", "Subtractive Box" simply do not appear unless set to a Large size.

( Separator added to more clearly show the buttons, without it both buttons don't appear )
[image]
[image]
[image]

( Without separator )
[image]
[image]

APEbbers, 2024-11-17:
Hi,

I've looked into it. apparently there is a maximum of 13 columns. There is no obvious parameter in PyQtRibbon, but i've now set that everything beyond 13 columns will added in the overflow panel.

SpeeQz1, 2024-11-17:
@APEbbers Tried the latest Develop build with the changes and I still seem to have the same issues.

APEbbers, 2024-11-17:
If you set the maximum number of columns to 6. Do you see then a button like this?
[image]

and if so, are the buttons then in the extra menu?

SpeeQz1, 2024-11-17:
Yep the overflow button does appear. Although it seems to not appear for the default value (mine was 0).

APEbbers, 2024-11-17:
And if you set the number of columns to 12?

I have it at 0 and it shows this (internally this is 13 or more):
[image]

SpeeQz1, 2024-11-17:
Still is missing.
[image]
[image]

APEbbers, 2024-11-17:
I've set the maximum to 10. In my virtual Linux, the buttons are now showing. Can you give it a try?

SpeeQz1, 2024-11-17:
[image]

SpeeQz1, 2024-11-18:
Hopefully the default would be changed as subtractive loft is missing or the bug is somehow fixed for the maximum number of columns.

APEbbers, 2024-11-18:
If you click on the arrow, button, is the missing button there?

APEbbers, 2024-11-18:
I think I've found it. There were some strange size policies needed for panels.
I've tested it on both windows and linux mint (VirtualBox) with FC1.0 QT6.

Can you try if it also works on your side?

SpeeQz1, 2024-11-18:
Seems to have worked although now the additive and subtractive pipes are separated by a separator.

[image]

APEbbers, 2024-11-18:
I had to add a separator between the boolan and the smaller icons. After that, it is it was intended again
[image]

closed 2024-11-18 by APEbbers