APEbbers/FreeCAD-Ribbon #111: Ribbon layout menu crashesh

SpeeQz1, 2024-11-18:
When attempting to open the Customize... -> Ribbon layout it would crash and give the following error:

```sh
21:52:34  "Empty widget item in QGridLayout 'gridLayout_8'."
21:52:34  Uncaught exception:
   File "/home/user/.local/share/FreeCAD/Mod/FreeCAD-Ribbon-Develop/./FCBinding.py", line 704, in loadDesignMenu
    LoadDesign_Ribbon.main()
  File "/home/user/.local/share/FreeCAD/Mod/FreeCAD-Ribbon-Develop/./LoadDesign_Ribbon.py", line 2552, in main
    Dialog = LoadDialog().form
             ^^^^^^^^^^^^
  File "/home/user/.local/share/FreeCAD/Mod/FreeCAD-Ribbon-Develop/./LoadDesign_Ribbon.py", line 242, in __init__
    MenuName = CommandInfoCorrections("menuText").replace("&", "")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/.local/share/FreeCAD/Mod/FreeCAD-Ribbon-Develop/./Standard_Functions_RIbbon.py", line 518, in CommandInfoCorrections
    CommandInfo = Command.getInfo()
                  ^^^^^^^^^^^^^^^

AttributeError: 'NoneType' object has no attribute 'getInfo'
Traceback (most recent call last):
  File "/home/user/.local/share/FreeCAD/Mod/FreeCAD-Ribbon-Develop/./FCBinding.py", line 704, in loadDesignMenu
    LoadDesign_Ribbon.main()
  File "/home/user/.local/share/FreeCAD/Mod/FreeCAD-Ribbon-Develop/./LoadDesign_Ribbon.py", line 2552, in main
    Dialog = LoadDialog().form
             ^^^^^^^^^^^^
  File "/home/user/.local/share/FreeCAD/Mod/FreeCAD-Ribbon-Develop/./LoadDesign_Ribbon.py", line 242, in __init__
    MenuName = CommandInfoCorrections("menuText").replace("&", "")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/.local/share/FreeCAD/Mod/FreeCAD-Ribbon-Develop/./Standard_Functions_RIbbon.py", line 518, in CommandInfoCorrections
    CommandInfo = Command.getInfo()
                  ^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'getInfo'
```

Usually before it may give an error or two but it would open and it would function normally without issues.

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.29 (Disabled)
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * CurvedShapes 1.0.13
  * FreeCAD-Ribbon-Develop 1.4.0
  * SearchBar 1.0.2
  * Backups
  * OpenTheme 2024.9.1
```

APEbbers, 2024-11-18:
Yes, i'm currently working on that part of the script. it's in progress

closed 2024-11-18 by APEbbers

SpeeQz1, 2024-11-18:
What about the missing "Create Sketch" button?

APEbbers, 2024-11-18:
That's actually what i'm working on! In FC, the commanddata for the sketch button and the datum button have the same data. I'm now implementing a function to correct command data

APEbbers, 2024-11-18:
It should be fixed now.
[image]

SpeeQz1, 2024-11-18:
> It should be fixed now. [...]

Still not getting it to appear. Removed the config files and also the addon from the `Mod` folder but no changes.

[image]

APEbbers, 2024-11-18:
The standard config file had an still error:
[image]
 is changed to:
[image]

SpeeQz1, 2024-11-18:
And it appears to be fixed.

[image]