APEbbers/FreeCAD-Ribbon #116: Feature request: Allow the user to set a universal default icon size and select the "icon only" checkbox for all panels of all workbenches.

raindropsfromsky, 2024-11-19:
The **Ribbon Design** dialog allows us to change the icon size for each panel of each workbench.
This is a fantastic feature.

However, I prefer the following settings for all workbenches:
1. All icons -> small size
2. No text (labels) for any workbench or panels.

So I had to select each workbench and then changing the icon size to small, and selecting the "Icons only" checkbox.
This was too tedious!

Please consider that a vast majority of FreeCAD users would like these settings as default.
Only novices will prefer to display text, or large icons at some places. 
But within a few days, they would also want to change over to the expert mode (small icons, no text.)

Therefore, the settings dialog should have two universal default settings: 
1. Small icons
2. No text (labels)

This should reset the size of all icons to "small", and turn off all labels in all workbenches and panels.

After that, if the user wants, he can always select a particular workbench and panel, and change the settings for specific icon(s).

This will save a lot of labor for the vast majority of FreeCAD users.

APEbbers, 2024-11-19:
In the next release the scripts is optimized to help with that. in the release after that this is planned for the design menu

raindropsfromsky, 2024-11-19:
Could you please share the link to the development version?
I can check it out and give advance feedback.

Thanks in advance!

APEbbers, 2024-11-19:
of course!

[https://github.com/APEbbers/FreeCAD-Ribbon/tree/Develop](url)

raindropsfromsky, 2024-11-19:
> https://github.com/APEbbers/FreeCAD-Ribbon/tree/Develop

I copied and pasted the link. It goes to the same "code" page of GitHub, where [v1.3.16](https://github.com/APEbbers/FreeCAD-Ribbon/releases/tag/v1.3.16) is listed as the latest version. (It was released 2 weeks ago.)

But I installed Ribbon WB using the Addon Manager, and that version is the latest (1.3.17.2)!
[image]

So which is the latest development version, and where is it available?

closed 2024-11-20 by APEbbers