APEbbers/FreeCAD-Ribbon #118: The Ribbon design dialog does not reflect all panels of the Surface workbench correctly

raindropsfromsky, 2024-11-19:
The **Ribbon design** dialog does not reflect all panels of the Surface workbench correctly, as shown in the screenshot below:
[image]

The actual workbench has many panels (marked "1").
In fact, some of the panels spill over on the right side of the screen (marked "2").

But the **Ribbon design** dialog shows only three panels.

As a result, we cannot customize the ribbon for this workbench.

**Desired:** The **Ribbon design** dialog should shows all panels correctly.

APEbbers, 2024-11-19:
you are showing the FEM workbench in the background. Surface and TechDraw WB are ok in my side.

closed 2024-11-19 by APEbbers

raindropsfromsky, 2024-11-19:
> This seems an cache issue. It is as if the wb is not properly activated. Sometimes it helps to set the wb to loaded at startup. I have seen this before and changed the code. So I'm closing it for now and mark it as fixed in develop branch.

I wonder if the other bug I reported also is because of that same reason (when I create a sketch from PD WB, FC switches to Part WB rather than to the Sketcher WB.)

raindropsfromsky, 2024-11-19:
The current version does not appear in the Preferences dialog (in the **Available Workbenches** list) at all
[image]