APEbbers/FreeCAD-Ribbon #121: Feature request: Allow user to turn off labels for workbenches

raindropsfromsky, 2024-11-19:
The user cannot turn off labels for workbenches.
As a result, the queue of the WB buttons runs off the screen.
This makes it difficult to select a workbench from the right side of the screen.

[image]

**Desired:**
1. Allow user to turn off the text labels for WB buttons
2. Provide a tooltip for these buttons (sometimes, the user may not recognize a rarely used WB)

APEbbers, 2024-11-19:
Hi,

With the current PyQtPackage this is not possible. In the future i will make my own version of it but that is more long term

closed 2024-11-19 by APEbbers

raindropsfromsky, 2024-11-19:
While fooling around with FC, I found this:
[image]

Can it be used to turn off the labels of WB buttons?

APEbbers, 2024-11-19:
Hi,

Thanks but sadly no. the problem is not with FreeCAD, but the PyQtRibbon package itself. It is a python package that creates the ribbon. The problem is that it heavily depends on the text being in the tabs. That's why I have to change the package itself.

I've tried to remove the text before, but then the ribbon chrashes. So it will be solved at a later point.