APEbbers/FreeCAD-Ribbon #122: The "Ribbon design" dialog should list the workbenches in the same order as the workbench toolbar

raindropsfromsky, 2024-11-19:
There is a major difference between the two listings. So finding the same workbench in both places is extra effort.
[image]

**Desired:** 
Let the "Ribbon design" dialog list the workbenches in the same order as the workbench toolbar

APEbbers, 2024-11-20:
I haven't found the cause yet. Reordering seems to reset the cache somehow. Also it can help to set the workbench which has problems to autoload on startup.

To solve this issue, I'm going to look in the serialize function of the SearchBar addon. that addon stores the data in a file.
That technique can be a solution for this as well for the loading time of the design menu.

hasecilu, 2024-11-20:
Did you installed the first bunch of external WB at once and then Curves WB?

APEbbers, 2024-11-20:
Hi,

I will order them alphabetically. That should solve this issue.

closed 2024-12-23 by APEbbers