APEbbers/FreeCAD-Ribbon #130: The ribbon panels do not match the FEM workbench

raindropsfromsky, 2024-11-20:
The ribbon panels do not match the FEM workbench.

[image]
As a result, the ribbon for the FEM WB is not controlled.

Reporting for Ribbon WB 1.4.1 with FC 1.1.0dev on Windows 11

APEbbers, 2024-11-20:
Hi,

in the develop branch, i've changed the timer to 500. Does that help?

APEbbers, 2024-11-20:
Does it help if you set the FEM workbench to autoload on startup?

hasecilu, 2024-11-20:
It seems not, also with latest changes now Ribbon is not loading

```
FreeCAD Ribbon: https://github.com/APEbbers/FreeCAD-Ribbon
wb Part Design not loaded
Uncaught exception:
   File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 776, in onUserChangedWorkbench
    self.onWbActivated()
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 797, in onWbActivated
    Gui.activateWorkbench(workbench.name())
                          ^^^^^^^^^^^^^^^^
  File "<string>", line 96, in name

AttributeError: 'PartDesignWorkbench' object has no attribute '__Workbench__'
Traceback (most recent call last):
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 776, in onUserChangedWorkbench
    self.onWbActivated()
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 797, in onWbActivated
    Gui.activateWorkbench(workbench.name())
                          ^^^^^^^^^^^^^^^^
  File "<string>", line 96, in name
AttributeError: 'PartDesignWorkbench' object has no attribute '__Workbench__'
CRITICAL:pyqtribbon_local.logger:Uncaught exception:
   File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 776, in onUserChangedWorkbench
    self.onWbActivated()
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 797, in onWbActivated
    Gui.activateWorkbench(workbench.name())
                          ^^^^^^^^^^^^^^^^
  File "<string>", line 96, in name

AttributeError: 'PartDesignWorkbench' object has no attribute '__Workbench__'
Traceback (most recent call last):
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 776, in onUserChangedWorkbench
    self.onWbActivated()
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 797, in onWbActivated
    Gui.activateWorkbench(workbench.name())
                          ^^^^^^^^^^^^^^^^
  File "<string>", line 96, in name
AttributeError: 'PartDesignWorkbench' object has no attribute '__Workbench__'
wb Part Design not loaded
Uncaught exception:
   File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1493, in __init__
    ribbon = ModernMenu()
             ^^^^^^^^^^^^
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 253, in __init__
    self.onUserChangedWorkbench()
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 776, in onUserChangedWorkbench
    self.onWbActivated()
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 797, in onWbActivated
    Gui.activateWorkbench(workbench.name())
                          ^^^^^^^^^^^^^^^^
  File "<string>", line 96, in name

AttributeError: 'PartDesignWorkbench' object has no attribute '__Workbench__'
Traceback (most recent call last):
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1493, in __init__
    ribbon = ModernMenu()
             ^^^^^^^^^^^^
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 253, in __init__
    self.onUserChangedWorkbench()
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 776, in onUserChangedWorkbench
    self.onWbActivated()
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 797, in onWbActivated
    Gui.activateWorkbench(workbench.name())
                          ^^^^^^^^^^^^^^^^
  File "<string>", line 96, in name
AttributeError: 'PartDesignWorkbench' object has no attribute '__Workbench__'
CRITICAL:pyqtribbon_local.logger:Uncaught exception:
   File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1493, in __init__
    ribbon = ModernMenu()
[... 100 lines cut ...]
  File "<string>", line 96, in name

AttributeError: 'SketcherWorkbench' object has no attribute '__Workbench__'
Traceback (most recent call last):
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 797, in onWbActivated
    Gui.activateWorkbench(workbench.name())
                          ^^^^^^^^^^^^^^^^
  File "<string>", line 96, in name
AttributeError: 'SketcherWorkbench' object has no attribute '__Workbench__'
CRITICAL:pyqtribbon_local.logger:Uncaught exception:
   File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 797, in onWbActivated
    Gui.activateWorkbench(workbench.name())
                          ^^^^^^^^^^^^^^^^
  File "<string>", line 96, in name

AttributeError: 'SketcherWorkbench' object has no attribute '__Workbench__'
Traceback (most recent call last):
  File "/home/uli/.local/share/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 797, in onWbActivated
    Gui.activateWorkbench(workbench.name())
                          ^^^^^^^^^^^^^^^^
  File "<string>", line 96, in name
AttributeError: 'SketcherWorkbench' object has no attribute '__Workbench__'
8.47128 MainWindow.cpp(1721): update actions
'PartDesignWorkbench', 1
```

APEbbers, 2024-12-03:
> The recent changes that fixed #77 seems to have affected on the FEM "Loads and Constraints custom", as mentioned this was fixed sometime ago but there is a regression on `Develop`
> 
> ![image](https://private-user-images.githubusercontent.com/53124818/392062264-c28448a0-a61b-4737-a811-63c8f9a86350.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMyNjMwMjAsIm5iZiI6MTczMzI2MjcyMCwicGF0aCI6Ii81MzEyNDgxOC8zOTIwNjIyNjQtYzI4NDQ4YTAtYTYxYi00NzM3LWE4MTEtNjNjOGY5YTg2MzUwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAzVDIxNTIwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAwZDEzOGZmODNiNWI4MWM4OWQyNTZkNjFlODZiNjIwNjJmNzNkMzkzY2M5M2VmZmY3MTNjNTQ5YTMxNzI1NTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.JYSCU-NqpN9kiYItO8mvtx39JSVIj8c9PaTwPll0Kk0)

This should now be solved again.

closed 2024-12-23 by APEbbers