APEbbers/FreeCAD-Ribbon #131: The labels for panels are too long and complicated in Exploded Assembly WB, which spoils the alignment of toolbar buttons

raindropsfromsky, 2024-11-20:
The labels for panels are too long and complicated in Exploded Assembly WB (marked "1").
[image]

As a result, the section becomes too wide, with too few buttons in it.
Consequently, these buttons appear to be scattered (marked "2")

**Suggestions:** 
1. Remove "ExplodedAssembly" at the beginning
3. Remove "tools" at the end

Both these parts are redundant!

closed 2024-11-20 by APEbbers

raindropsfromsky, 2024-11-23:
Well, a custom toolbar should not be needed as a solution for this issue: It would be enough to clean up the labels for this toolbar as suggested, IMHO.

APEbbers, 2024-11-23:
I understand. But is not an issue for the Ribbon, but for the workbenches them self. 
The reason that I hesitate to create custom labels is:
 - if I make custom labels, I have to do that for every possible workbench which is a lot of maintenance
 - People preferer different names. So it is never right for everyone. It is better to let people decide for them selves.
 - Naming of toolbars is primarily the responsibility of the WB maintainer. They know their design philosophy best.
 
It is a known issue with FreeCAD but there is not a definite solution yet.
I have placed the issue on the roadmap for the long term. (See my wiki in this repo)

raindropsfromsky, 2024-11-23:
Ah, I get it!
Thanks!