APEbbers/FreeCAD-Ribbon #134: The font size should not vary widely

raindropsfromsky, 2024-11-20:
The UI of the Mirror WB uses a wild variation in the fonts (see the difference between labels in the screenshot below).
[image]

Here is another screenshot that shows that the font size has a wide variation, which is not based on the hierarchy of information.
[image]

The only criteria seems to fit the text in the available length. So longer labels have tinier font.

**Desired:** 
In a UI, a main label is supposed to have the largest font.
As we go down the hierarchy of the UI elements, the font should become progressivly smaller.

The longer titles can be made shorter in most cases.
For example, the name of any panel begins with the name of the WB it belongs to.
Probably this is done to facilitate making a custom toolbar, where the user has collected tool buttons from multiple WBs.
But this would not be required by most users.
So this odd requirement spoils the UI for the entire WB.

APEbbers, 2024-11-20:
Hi,

You can change the size of the tabbar text, by changing the icon size for
the tabbar
But to be honest it seems quite normal to me, that the panels have a
smaller font than the tabs.

I'm trying to avoid settings per wb, because the is often a personal
preference and thus results on a lot of maintenance.

kind regards

closed 2024-11-22 by APEbbers

raindropsfromsky, 2024-11-23:
Well, it is ok if the tabs have the largest font. But they are larger than the next largest font by a huge margin. 
That leaves the other text too tiny. 
A difference of two points should be enough. 

Secondly, as the screeenshot shos, the menu system is not given a progressively decreasing font size.
(Menu > submenu > sub-submenu)

Thus, there are two separate issues here.

APEbbers, 2024-11-23:
I've recently updated  the ribbon. I've changed the default values to smaller sizes. When you press reset, it should look like this:
[image]

The second issue is a bit strange for me, because I don't have different sizes between the menus:
[image]

I'm mostly working with Windows 11, FC 1.1.0dev and QT6. I'll have a look if I can find a cause for it.

raindropsfromsky, 2024-11-23:
Thanks for changing the font size for the top bar (WB toolbar).

The second issue could be because of my custom scale factor, which I have set to 115%.
The reason is as follows: My Aspire Lite AL15-52 laptop has FHD monitor, and its recommended scale factor is 125%.
But at this scale factor, its icons in the Task Manager become too large.
At 100% scale factor, they appear too small.
So I found the sweet spot at 115%.

I have faced similar issue with FreeCommander also.