APEbbers/FreeCAD-Ribbon #138: Ribbon Preferences window is too tall above 1080 pixels

julianjc84, 2024-11-20:
The Ribbon Preferences window is too tall above 1080 pixels displays
and not able to be resized.

hasecilu, 2024-11-21:
At the moment of opening not all elements are properly displayed (Hyprland) but now the window can be resized!, Do you have latest version?

[image]

julianjc84, 2024-11-21:
1.4.2
the preference window can be made taller, but the shortest allowed is over 1651pix
[image]

hasecilu, 2024-11-21:
Normally FreeCAD preference pages automatically have a scrollbar when it's needed but since this is a standalone window the scroll have to be implemented on the UI file.

APEbbers, 2024-11-21:
Hi,

I've added a scrollbar feature in both the layout dialog as in the preference dialog

kind regards

closed 2024-11-21 by APEbbers