APEbbers/FreeCAD-Ribbon #152: The toolbar for the Techdraw WB does not show many of the panels listed in the Ribbon Design UI.

raindropsfromsky, 2024-11-23:
The toolbar for the Techdraw WB does not show many of the panels listed in the Ribbon Design UI.

[image]

**Desired:**

All panels listed in the Ribbon Design dialog should be shown in the Ribbon.

APEbbers, 2024-11-23:
I've had that a couple of times. It will be solved when you set the TechDraw Workbench autoloaded.

raindropsfromsky, 2024-11-24:
Should I raise this bug at FC GitHub?

APEbbers, 2024-11-24:
This can be a bug where FreeCAD is not properly loading the WB. So you may raise it at FC Github.
I'm looking into it however.

APEbbers, 2024-11-25:
Hi,

I suspect that the test workbench is the cause of the workbenches not loading properly. I've currently the test workbench disabled and no problems so far. Perhaps you can give it a try as well?

raindropsfromsky, 2024-11-26:
I do not have a WB called "Test" in the WB toolbar.
I checked the Addon Manager also. It does not have any Test WB.

[image]
(Note that I have selected "any workbench" filter.)

APEbbers, 2024-11-26:
it is not an normal add-on. You have to look under preferences -> workbenches. There you will see a wb test:
[image]

raindropsfromsky, 2024-11-27:
The **Test framework** was already unchecked for me from the start. 
Yet I faced the issue.

hasecilu, 2024-12-03:
Similar to last comment on #130 "Dimensions custom" panel is empty.
[image]

closed 2024-12-23 by APEbbers