APEbbers/FreeCAD-Ribbon #155: When I changed the Addon Manager setting of some WBs to "Auto-load", their button size got reset in the Ribbon WB

raindropsfromsky, 2024-11-24:
I had previously gone through all my selected WBs, and changed all buttons to "small" using the Ribbon WB's Layout dialog.
It was working fine till today.

But I changed the Addon Manager's settings for some of these WBs to "Auto-load".
And now I find that a lot of buttons in multiple WBs have reverted to "large".

I will have to again repeat the mammoth exercise of setting all buttons to "small"!

**Desired:**
1. The "auto-load setting should not interfere with button size.
2. Please make all buttons "small" by default, as most users need that setting only.
3. Please provide a checkbox to force all buttons to "small" in all WBs, all panels.

raindropsfromsky, 2024-11-24:
Should I raise this bug at FC GitHub?

APEbbers, 2024-11-24:
No, this is the Ribbon related.

closed 2024-12-18 by APEbbers