APEbbers/FreeCAD-Ribbon #156: Large buttons do not follow the Ribbon Preference settings

raindropsfromsky, 2024-11-24:
I wanted all buttons in all WBs to be "small".
But for that, I have to select all WBs one by one, then select each panel one by one, and then change the size of the buttons to "small".
This is too tedious.

Unfortunately, I had to repeat this exercise three times.

I lost the settings today when I changed the settings of some WBs to "Auto-load".

So, rather than repeating the whole mammoth exercise all over again, I changed all button sizes to 25 in Ribbon WB's **Preferences** section. 

Logically, since all buttons are of the same size, I expected them to reflow into a grid.

But the Ribbon UI does not follow that setting, as shown below:
[image]

- The large buttons have not resied to 25 (to match the "small" buttons)
- The large buttons have not reflowed into a grid: Each large button occupies a separate column.
   (This may be by design, though)

APEbbers, 2024-11-24:
Hi,

This is indeed by design. I'm planning to extend the layout dialog with extra functions like:
- set all buttons to one size per wb
- set all buttons to one size per panel
- set all buttons to one size for all
- import/ export control settings from backups

For the moment, you can run a script to set all controls to one size.
[image]

Another option is to open "RibbonStructure.json" in notepad and do a search and replace.
Make sure that "small" is lower case.

raindropsfromsky, 2024-11-24:
That is wonderful!

IMHO we do not need a separate setting per WB.
(That again would force the user to run the script N times, where N is the number of installed WBs.)
It is fine to have just one size for all WBs at one go.

BTW if I install a new WB later, I have to run the script once again, right?

Which of these scripts sets all icons to a uniform size?
(I could not make that out from the names.)

APEbbers, 2024-11-24:
I'm going to add all three, so the user can choose.
Yes you have to run the  script again,. That is if you want other sizes than small. if the wb is not present in the json file, all will be small anyway.

It is the script "CreateDefaultRibbonStructure.py". In the manual [(https://github.com/APEbbers/FreeCAD-Ribbon/wiki)](url) and the script itself are instructions on how to use it.

You can also open "RibbonStructure.json" in notepad and do a search and replace. make sure to write "small" as lowercase.

raindropsfromsky, 2024-11-24:
> You can also open "RibbonStructure.json" in notepad and do a search and replace. make sure to write "small" as lowercase.

I did this. Although I had difficulty in making just 7 buttons in three WBs to small size, when I searched for "large", I found 52 instances in the json file! I changed them all with "small", saved the file and restarted FC. Now all ribbons are fine.

closed 2024-11-25 by APEbbers