APEbbers/FreeCAD-Ribbon #176: Small errors in padding and buttons

MarcelP102, 2024-11-29:
Hi, I noticed some places where there seems to be to much padding.

Below the ribbon:
[image]

At the right side of the buttons (large, medium and small) there is always to much space, making the button off-center. I think you always reserve space for the dropdown button, also when they are not used. I would advise to only make padding when there is a dropdown. And also make the padding at both sides smaller. See here compared to word:
[image]
vs
[image]

Left and right side of the separator:

Good:
[image]

Bad:
[image]

Icons become smaller when hovering the mouse:
[image]

Contour of button is not fully displayed:
[image]

APEbbers, 2024-11-29:
Hi,

Thanks for the suggetions! Regarding the padding, the Ribbon buttons are based on QToolbuttons, which centers the icons. It is not possible to left align an icon on a QToolbutton with CSS. I've planned to create a custom class for the buttons to be able to align the icons. 

I'm working on the padding below the ribbon. On my system, from size 30 onwards, it seems ok. Smaller and a "padding" appears. i'm looking into it.

Regarding the disappearing outline: which sizes have you set? below is a picture with size 20 for small buttons from my system:
[image]

Which version of the ribbon do you have installed and which OS version are you using? (I'm checking because I've tested it with both Windows 11 and Linux mint and it seems fine)

Good idea to use a background instead of a border. I'll implement that for the next release.

MarcelP102, 2024-12-01:
Hi,

I use Windows 11, Freecad 1.0, Ribbon 1.4.10 and these settings:
[image]

Looking forward to the improvements :)

MarcelP102, 2024-12-01:
I now got the development branch working, already a big improvement with the icons padding, great work!

closed 2024-12-23 by APEbbers