APEbbers/FreeCAD-Ribbon #182: Option to enter the number of rows and quit the bottom panel names.

dgrabulosa, 2024-12-01:
With all small icons, there are 3 rows in the ribbon, and a lot of space in the end it can fill if y can select 2 rows for example.
The other ting is the panel names, if I can turn off I can reduce the width of the ribbon and add mote space to the main screen.
[image]

MarcelP102, 2024-12-01:
It wasn't clear for me either but changing the icons to medium in the Ribbon layout - Ribbon design tab makes in into 2 rows. So:
Large = 1 row
Medium = 2 rows
Small = 3 rows

dgrabulosa, 2024-12-01:
The rows control is important to optimize the available space: with 3 rows, the ribbon leaves near 50 % without icons. In my part design workbench.
If I can configure 2 rows and hide section titles, I can recovery near 30%-40% of the space occupied with the icons of the ribbon.

APEbbers, 2024-12-30:
Hi,

I've compared several programs with a ribbon and they all have a panel title. Also I believe thatpanel titles are an important part of a ribbon design.  The are used to group different type of functions like with toolbars.
Combined with the fact that since the latest release of the ribbon, the titles are much shorter, the titles remain present.

kind regards,

P.S.: below two examples. (Outlook and SolidEdge)
[image]
[image]

closed 2024-12-30 by APEbbers