APEbbers/FreeCAD-Ribbon #183: Request: Option to hide a icon

MarcelP102, 2024-12-01:
Would be nice if you uncheck any icon size the icon would not be displayed. Right now they keep showing. Or add an option to hide the icon. This would be easier than making custom panels.

[image]

MarcelP102, 2024-12-01:
For now I use the Tools-Customize function of freecad to remove tools form a toolbar.

hasecilu, 2024-12-01:
You could create a custom with only the tools you need. Or even you could edit the JSON file.

APEbbers, 2024-12-29:
Hi,

It is added to the develop branch:
[image]

kind regards

closed 2025-01-08 by APEbbers