APEbbers/FreeCAD-Ribbon #184: Request: Set as default for all workbenches

MarcelP102, 2024-12-01:
When designing ribbons some workbenches have the same panels, like for example the panel 'Tools'. It would be nice to have an option to change the design (small-medium-big) to all workbenches that have that panel.

So I suggest to add an 'set as default' to set the icon size of all workbenches using this panel.

[image]

hasecilu, 2024-12-01:
I noticed that those panels are clones on every distinct WB and thus act as different panels, so you have to redo your changes again and again, I think would be better if instead of having N instances of the same panel a single one is created and reused, in case a different format is needed then another one should be created, "Tools", "Toolsv2", etc.

Look how many panels are created!
[image]

APEbbers, 2024-12-01:
No it is a bug. It should only show only one of each panel.

might also be an translation issue. In English it is fine. I'm on it.

hasecilu, 2024-12-03:
- Exclude panels tab is not longer with duplicated entries, so moving them to exclude list works
- Initial setup now it's populated.

MarcelP102, 2024-12-16:
Works now as expected in dev branch

closed 2024-12-23 by APEbbers