APEbbers/FreeCAD-Ribbon #189: Ribbon not loading

Nekhaevalex, 2024-12-07:
### Description
After installation and restart ribbon won't load (may be seen on an attached image)
[image]

### Report message:
```
23:50:21  Activating Ribbon Bar...
23:50:23  FreeCAD Ribbon: 
23:50:23  QWidget::setMaximumSize: (Ribbon/QDockWidget) Negative sizes (16777215,-20) are not possible
```

### Installation info:
OS: macOS 15.1.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * SearchBar 1.0.2
  * FreeCAD-Ribbon 1.4.14
  * OpenDark 2024.3.13
  * sheetmetal 0.6.1

APEbbers, 2024-12-07:
Hi,

I've updated the code so that there won't be any negative value possible.  Can you test if version 1.4.15 solves the problem?

Nekhaevalex, 2024-12-07:
Hi!

The message in Report panel doesn't appear anymore. However, the ribbon widget is still a grey rectangle without any buttons.

APEbbers, 2024-12-07:
Hi,

I'm trying to setup a test enviroment for MacOS. Until then, I'm a little blind. I did however make another change.
Can you perhaps try version 1.4.15.1? 

If the ribbon is still not showing, can you enable all messages in the report view? (right click -> options -> display message types)
I hope that you will see more information about what is happening.

kind regards

Nekhaevalex, 2024-12-07:
Hi!

Still empty frame. Report log seems to be ok:
```
12:35:01  Activating Ribbon Bar...
12:35:01  Init:      Initializing /Users/alex/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./... done
12:35:01  All modules with GUIs using InitGui.py are now initialized
12:35:01  All modules with GUIs initialized using pkgutil are now initialized
12:35:01  Init: Loading FreeCAD GUI
12:35:01  Init: Running FreeCADGuiInit.py start script... done
12:35:03  Init: Activating default workbench PartDesignWorkbench
12:35:03  Loading Sketcher module... done
12:35:03  Loading PartDesign module... done
12:35:03  Loading GUI of Material module... done
12:35:03  Loading GUI of Part module... done
12:35:03  Loading GUI of Sketcher module... done
12:35:03  Loading GUI of PartDesign module... done
12:35:03  FreeCAD Ribbon: https://github.com/APEbbers/FreeCAD-Ribbon
12:35:03  Init: Showing main window
12:35:03  Main window restored
12:35:03  Show main window
12:35:03  Toolbars restored
12:35:03  3Dconnexion framework not found!
12:35:03  Init: Entering event loop
12:35:03  Init: Executing event loop...
12:35:03  Init: Processing command line files
```

Worufu3D, 2024-12-08:
Hi, same issue in Windows 10, Ribbon 1.4.15.2 can be loaded only manually from view selection.

Here the log report:
19:13:05  OpenGL version is: 4.6 (4.6.0 NVIDIA 560.94)
19:13:08  Run Gui init script
19:13:08  Init: Running FreeCADGuiInit.py start script...
19:13:08  Init:   Searching modules...
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\AddonManager... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Assembly... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\BIM... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\CAM... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Draft... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Fem... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Help... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Idf(InitGui.py not found)... ignore
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Import... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Inspection... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Material... done
19:13:08  Loading Material module... done
19:13:08  Module: Part
19:13:08  Loading Part module... done
19:13:08  Loading Measure module... done
19:13:08  Loading GUI of Measure module... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Measure... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Mesh... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\MeshPart... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\OpenSCAD... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Part... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\PartDesign... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Plot(InitGui.py not found)... ignore
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Points... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\ReverseEngineering... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Robot... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Show(InitGui.py not found)... ignore
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Sketcher... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Spreadsheet... done
19:13:08  Loading GUI of Start module... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Start... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Surface... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\TechDraw... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Test... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Tux... done
19:13:08  Init:      Initializing C:\Program Files\FreeCAD 1\Mod\Web(InitGui.py not found)... ignore
19:13:09  Activating Ribbon Bar...
19:13:09  Init:      Initializing C:\Users\Wolf\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\... done
19:13:09  Init:      Initializing C:\Users\Wolf\AppData\Roaming\FreeCAD\Mod\PieMenu\.\... done
19:13:09  Init:      Initializing C:\Users\Wolf\AppData\Roaming\FreeCAD\Mod\SearchBar\.\... done
19:13:09  All modules with GUIs using InitGui.py are now initialized
19:13:09  All modules with GUIs initialized using pkgutil are now initialized
19:13:09  Init: Loading FreeCAD GUI
19:13:09  Ignoring unknown NoneWorkbench workbench found in user preferences.
19:13:09  Ignoring unknown TestWorkbench workbench found in user preferences.
19:13:09  Init: Running FreeCADGuiInit.py start script... done
19:13:14  Init: Activating default workbench PartDesignWorkbench
19:13:14  Init: Showing main window
19:13:14  Main window restored
19:13:14  Show main window
19:13:14  Toolbars restored
19:13:14  FreeCAD Ribbon: 
19:13:14  Loading Fem module... done
19:13:14  Loading GUI of Fem module... done
19:13:14  Loading Sketcher module... done
19:13:14  Loading PartDesign module... done
19:13:14  Loading GUI of Material module... done
19:13:14  Loading GUI of Part module... done
19:13:14  Loading GUI of Sketcher module... done
19:13:14  Loading GUI of PartDesign module... done
19:13:14  Loading TechDraw module... done
19:13:14  Loading TechDrawGui module... done
19:13:14  Init: Entering event loop
19:13:14  Init: Executing event loop...
19:13:14  Init: Processing command line files

[image]

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * FreeCAD-Ribbon 1.4.15.2
  * PieMenu 1.10.1
  * SearchBar 1.2.1

APEbbers, 2024-12-08:
Hi,

It took me a while but I got finally a test enviroment for MacOS working.  After testing, I think I found a cause and solution.
The ribbon is based on the PyQTRibbon package. As it turns out, the PyQTRibbon package based the ribbon on a QMenuBar. I've changed this to QWidget. On my system it seems to be working. 

(The issue is that the menubar is detached from the window in MacOS).

can you try by switching to the branch: https://github.com/APEbbers/FreeCAD-Ribbon/tree/189-ribbon-not-loading? and see if this one is working?

Thanks for your patience and help!

kind regards

Worufu3D, 2024-12-08:
> > Hi, same issue in Windows 10, Ribbon 1.4.15.2 can be loaded only manually from view selection.
> > Here the log report:
> 
> > ![immagine](https://private-user-images.githubusercontent.com/4172408/393632049-10e36d3e-3ebb-45f2-9567-284d49619bc1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM2ODM5MTIsIm5iZiI6MTczMzY4MzYxMiwicGF0aCI6Ii80MTcyNDA4LzM5MzYzMjA0OS0xMGUzNmQzZS0zZWJiLTQ1ZjItOTU2Ny0yODRkNDk2MTliYzEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMDhUMTg0NjUyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjhkNzVjN2ZjOGQwOTFhMmE4NTZiMWRkZTc2MjJmOWMwNWQyZmFiZjhlOTE0OTZhYzAyOTdhYTY3ODM3ODVlNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.u0mtMJVSHpIJY8RXx8D7yo-cMqUYtBMHa0CZEpQLREw)
> > OS: Windows 10 build 19045 Architecture: x86_64 Version: 1.0.0.39109 (Git) Conda Build type: Release Branch: (HEAD detached at 1.0.0) Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20 Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1 Locale: Italian/Italy (it_IT) Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion Installed mods:
> > 
> > * FreeCAD-Ribbon 1.4.15.2
> > * PieMenu 1.10.1
> > * SearchBar 1.2.1
> 
> Hi,
> 
> After you have activated the ribbon via views, on the next startup the ribbon is hidden again? Or does it stay visible? edit: I changed a line. Can you try again on the branch mentioned above?

Good work, with this branch Freecad remember Ribbon and it stays visible.

Nekhaevalex, 2024-12-08:
Hi all!

I tried branch https://github.com/APEbbers/FreeCAD-Ribbon/tree/189-ribbon-not-loading and it works!

Thank you so much!
[image]

APEbbers, 2024-12-09:
Hi,

Great to hear that the issues are solved. I've updated the main branch to version 1.4.16 with the update. It is best to switch back to the main branch for future updates.

kind regards

closed 2024-12-09 by APEbbers