APEbbers/FreeCAD-Ribbon #194: Cannot change icon size by editing either the JSON file or the script file

raindropsfromsky, 2024-12-11:
**Background:**
After trying to press F3 to hide the ribbon, FC showed a totally blank screen.
I could not recover normal screen.
So I had to re-install FC.
I also removed the Mod folder.

In the new installation, I wanted to change all icons to "small" size.
I tried to achieve this by editing the JSON file, which did not work.
Then I tried to edit the script file, which also did not work.

**Here are the details:**
First, I closed FC, and then opened the JSON file in NotePad++.
Then I carried out find-and-replace (all "large" with "small").
Now all instances of icon size have the new value "small", as shown below.
`"size": "small",`

I saved the JSON file, and then started FC. 
The ribbon does not show all icons with small size.

------------------

After failing to change the icon size, I tried to do this through the script file.

So I opened the script file **CreateDefaultRibbonStructure.py** in this path:
"C:\Users\UserName\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\Scripts\"

Again, I replaced all instances of "large" to "small".
After saving, I ran FC.
(I expect FC to run the script on its own, because there is no explicit way to run it.)

But when I started FC, there was no change.

raindropsfromsky, 2024-12-11:
**Update:**

Found the RibbonStructure.json in this path: 
C:\Users\UserName\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon

I did the same find-and-replace in this file, and got the results.

But this highlights the main issue: It is difficult for a user to fish for the correct file and do the editing.
It should be possible to set the icon size directly from the menu (please provide a codeless method).

APEbbers, 2024-12-11:
Hi,

In the next release, you can set all buttons to your preferred size from the layout dialog. This function is already implemented in my develop branch and will be available with version 1.5.

i hope to release version 1.5 this or next weekend.

kind regards

closed 2024-12-11 by APEbbers