APEbbers/FreeCAD-Ribbon #196: Missing icons in sketcher

MarcelP102, 2024-12-16:
[image]

APEbbers, 2024-12-16:
Hi,

That's probably because the data file is not generated yet. Can you test if it helps when you open the layout menu? The first time it will ask you to load the workbenches. When pressed "yes" a data file is generated. This is used as backup for the icons in the ribbon and for loading the layout menu faster

MarcelP102, 2024-12-16:
Yeah I already did that. But when I reload the workbenches in the ribbon design dialog I get this error:
[image]

[image]

Also another issue; the ribbon design dialog is behind the freecad main screen after reloading the workbeches.

APEbbers, 2024-12-16:
I'll have a look tonight. it is the normal behavior for these buttons. To cover this, the Icons, should be taken from the data file. But somehow this functionality is broken. At the moment it does work on my side, but I'll do some tests

MarcelP102, 2024-12-16:
Somehow the icons in sketcher now works correctly. But reloading the workbenches in the ribbon design dialog still give the same error

APEbbers, 2024-12-16:
Hi,

The issues should now be fixed. Regarding the missing Icons in the sketch wb, I couldn't reproduce the issue, but I keep it in mind.

kind regards

closed 2024-12-16 by APEbbers