APEbbers/FreeCAD-Ribbon #198: Links don't respect default browser

MarcelP102, 2024-12-17:
All the link for example the 'what's new?' link in the menu or the '?' link in the Ribbon layout all open the Edge browser instead of my default browser (firefox). 

Ps. the 'what's new?' link points to an invalid link (http://wiki/06-%E2%80%90-Change-log)

APEbbers, 2024-12-17:
Hi,

The links should work now. However, it is a bit strange that the links are opening in a non-default browser.
I used the native python package "webbrowser" on purpose because it should always open links in your default browser.

Looking on the internet, it can be that it is caused by the bad urls. Does it still opens in the wrong browser?

MarcelP102, 2024-12-18:
It's working now as expected, thanks.

closed 2024-12-18 by APEbbers