APEbbers/FreeCAD-Ribbon #207: Three bugs when I launch Ribbon WB's "Customize > Ribbon Layout" menu command

raindropsfromsky, 2024-12-23:
I tried out the Ribbon WB's **Customize > Ribbon Layout** menu.

But I see this error: **[BUG#1]**
[image]

After that, I see this screen: **[BUG#2]**
[image]

After that, I see this screen: **[BUG#3]**
[image]
I expected the "Reload the WBs" message to disappear after I select it once.

APEbbers, 2024-12-23:
can you try when you remove the RibbonStructure.json? and see what happens?

P.S. Bug 3 is not a bug, but just a warning that reloading can take a while. It is there permanently.

raindropsfromsky, 2024-12-24:
**Bug#2:**
I have not changed the paths this time, and I have never had any issue like this so far.
So it may not be related to the access rights.

Mine is FC 1.1..0 dev (not 1.0 Dev). Could that be the reason?

------------------------

The RibbonStructure.json is located at-
C:\Users\<UserName>\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon
I have "full control" (read, write, modify, execute) on the file and the folder.

I closed FC, and then moved the RibbonStructure.json file and two dat files to a local folder.
FC started normally.
I could launch the "**Customize > Ribbon Layout**" command without any issue.

---------------

**Bug#3:**

IMO this must not be permanently placed there.
After the user presses the **Refresh** button, the control should be removed.

(If the UI keeps showing him the same message, he concludes that his previous attempt failed.)

In FC itself, the Addon Manager provides an excellent example for handling the refresh:
It shows a progress bar and the current status:
[image]
[image]

It would be great if a similar UI is provided in Ribbon also.

APEbbers, 2024-12-24:
Bug#2: when you see this, what is reported in the report view? perhaps it is a specific workbench that causing this issue.

raindropsfromsky, 2024-12-24:
You mean bug#1? (Nonetype object)
I am not getting it any longer!

APEbbers, 2024-12-24:
OK. I'm parking this for the moment then. 

Regarding bug#3: I've tried this, but because the dialog was freezing until the very end. so I abandoned it. I'll keep it in mind though