APEbbers/FreeCAD-Ribbon #214: Add same panel proprieties like task panel: transparency, overly,....etc....

dgrabulosa, 2024-12-24:
I manage to obtain a clean panel distribution, but I have some errors.
- task panel appear that have same height as the ribbon.
- the ribbon panel generates an error If I try to click in the FreeCAD main icon.

Maybe if the ribbon panel have the same tools that have the task or tree pane in the header: transparency, overlay,...... can help.

OS: Windows 11 build, 22631
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Spanish/Spain (es_ES)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * FreeCAD-Ribbon 1.5.0.4

[image]

APEbbers, 2024-12-25:
Hi,

I created the custom overlay function, because the original overlay function is interfering with the ribbon.  The Ribbon is excluded from the overlay function. How did you get the ribbon transparant? That shouldn't be possible with the custom overlay function. Perhaps that's causing the error.

Regarding the task panel and tree, You can drag these to somewhere else. Click and hold the header, then you can drag the task panel.

APEbbers, 2024-12-25:
Hi,

I found a possible solution. Can you try my develop branch? You might have to set this option:
[image]

@hasecilu and @raindropsfromsky: Can you perhaps test as well this if this works? it might be a better solution for the problems with overlay and keep the native functionality instead

Kind regards,

hasecilu, 2024-12-25:
What exactly need to be tested? For what I see if you set the autohide option you are forced to click on the WB tab to show again the ribbon, should be on hover?

raindropsfromsky, 2024-12-25:
> if you set the autohide option you are forced to click on the WB tab to show again the ribbon, should be on hover?
I agree. The ribbon should appear on hover. Otherwise it will eat up too many clicks!

Also, what are the other features to be tested? Please do share!

APEbbers, 2024-12-25:
Hi,

I'd like to test if FreeCADs Overlay function causes any issue with the ribbon. When you click on the ribbon and then press F3, the ribbon should become transparant but not hidden when you have a part open. 

I've also fixed a bug in the PyQtRibbon package which was caused by hovering over the application button in overlay mode

raindropsfromsky, 2024-12-25:
Well, the selection status of the menu options is not clear:
[image]

Usually, we should see a tick, or a dot against the menu that is selected.

I like the idea of text wrap with large/medium buttons.
The small buttons do not have that much width, so they cannot have text wrap.

APEbbers, 2024-12-25:
Ok. I'll add dots at the end. But what happens when you click on the ribbon and press F3? (Or go to view->Docview overlay.
If it is not there you have to activate it first:
[image]

Maybe also a good idea to have some panels activated. You can then right click on them to enable the ribbon or other panels.

raindropsfromsky, 2024-12-25:
Actually the logic is not clear to me. 

1. What is "overlay" mode?
2. Can we have four different combinations of overlay and transparency modes?
    (on+on, off+on, on+off, off+off)
3. How is F3 supposed to interact with these two modes?

APEbbers, 2024-12-25:
In the latest commit I added a restore function.
If you put in a file named "OVERLAY_DISABLED" (no extension), the custom overlay function will be triggerd which restores the ribbon
[image]

> Actually the logic is not clear to me.
> 
> 1. What is "overlay" mode?
> 2. Can we have four different combinations of overlay and transparency modes?
>    (on+on, off+on, on+off, off+off)
> 3. How is F3 supposed to interact with these two modes?

You find the overlay function from FreeCAD here:
F3 triggers the first command.  The modes you previously mentioned are now hidden when the original overlay function is used.
The belong to my version.
[image]

hasecilu, 2024-12-25:
Mmm, OK I needed that last setting, after that pressing `F3` made switch to Overlay mode.

The ribbon no longer takes al horizontal space. If you try to move it to the top it comes back to normal mode.

[image]

raindropsfromsky, 2024-12-25:
I tried to press F3 and got some weird results.
The layout changes completely.

Before selecting the **Activate overlay handling** option, pressing F3 toggled the UI like this:
[image]
[image]
This is totally confusing!

After selecting the **Activate overlay handling** option, pressing F3 made the ribbon shrink a little, and after that, it disappeared completely, leaving the screen completely blank. 
[image]

This is terrible!
First of all, I have not really understood how this overlay feature will is supposed to work, or add value.

APEbbers, 2024-12-25:
Can you open a part by double clicking in texplorer? the ribbon should appear then

raindropsfromsky, 2024-12-25:
I restarted FC and the UI got restored.
But still I am not clear about what the overlay mode is supposed to achieve.

APEbbers, 2024-12-25:
It supposed to look like this:
[image]

raindropsfromsky, 2024-12-25:
@hasecilu I am unable to click on your image to see its magnified view.

APEbbers, 2024-12-25:
I can add this as an option with the comment "Use at your own risk"? and set the custom overlay function as default

hasecilu, 2024-12-25:
With overlay mode you are able to interact with the 3D view even when clicking above other panels and also the transparency to see the the 3D view, it's really useful with small display because sometines panels takes too much spaces.

Try reloading the page because I updated the image.

raindropsfromsky, 2024-12-25:
> It supposed to look like this:

No, I do not achieve that state at all. (Transparent toolbars over the canvas).
Instead, pressing F3 changes the panel layout totally!

hasecilu, 2024-12-25:
It also changed layout for me but manually restored, probably a FreeCAD thing

raindropsfromsky, 2024-12-25:
Only the mode is supposed to change with F3, and not the layout, right?

Secondly, why give F3 when the same mode is available from the menu system??

Thirdly, the menu system does not show "F3" shortcut:
[image]

APEbbers, 2024-12-25:
Thoes are the buttons from the custom overlay function. in the latest commit, they are hidden when The overlay function of FreeCAD is used.

raindropsfromsky, 2024-12-25:
I am using Ribbon 1.50.5, but it is totally buggy.
1. The menu system does nnot show the status of the overlay and transparency.
2. The layout changes whenever I press F3. Restoring it is a big task (I don't know how)

APEbbers, 2024-12-25:
Can you switch to the develop branch? the changes are there not in version 1.5.05.

raindropsfromsky, 2024-12-25:
> Thoes are the buttons from the custom overlay function. in the latest commit, they are hidden when The overlay function of FreeCAD is used.

I  am afraid I did not get that!

I have selected the **Activate overlay handling** option in the **Preferences**.

But I do not know how the menu system is supposed to work.
What are the steps to check this out?

raindropsfromsky, 2024-12-25:
The names of the branches have changed:
[image]
Which branch should I select?

raindropsfromsky, 2024-12-25:
I selected the "**origin/develop**" branch.
It produced this error:
[image]

APEbbers, 2024-12-25:
Not an error, but a warning. click "yes"

raindropsfromsky, 2024-12-25:
I updated the version to 1.6.x.
It has the same buggy behavior. (no transparency, the side panels change to undesired combo.)

APEbbers, 2024-12-25:
Did you enable the overlay function?
[image]

hasecilu, 2024-12-25:
Try `Shift+F3` for transparency.

APEbbers, 2024-12-25:
@hasecilu do think I should implement this as an option?

raindropsfromsky, 2024-12-25:
I toggled the overlay using the **Overlay > Toggle overlay** menu.

It has some change in the canvas area:
[image]
[image]

Pressing F3 haas the same effect.
Therefore I conclude that the menu should display F3 hotkey against this menu option.

Toggling the transparency using the **Overlay > Toggle transparency** menu does not change anything.

raindropsfromsky, 2024-12-25:
Pressing SHFT+F3 behaves exactly like pressing F3.

BTW after I click in the ribbon area, and then press F3 (or SHFT+F3), then the focus is lost from the ribbon area.
If I press F3 (or SHFT+F3) again, it has no effect.
I have to click in the ribbon area and then press F3 (or SHFT+F3).

I expected that the focus will remain in the ribbon area, so F3 and SHFT+F3 keeps toggling the overlay state without needing clicking in the ribbon area repeatedly.

APEbbers, 2024-12-25:
In this picture below you see what the keys should do: (This is FreeCAD itself)
[image]
 If you press F4/shit F4 instead. All panels will be triggered.

I just found out that you need to set the theme correctly:
[image]

raindropsfromsky, 2024-12-25:
I prefer opendark theme.
[image]

APEbbers, 2024-12-25:
Just tested it with those settings and it should work.
[image]

hasecilu, 2024-12-25:
> @hasecilu do you think I should implement this as an option?

**I think it's more useful to make the ribbon to show automatically when hovering than trying to make it work on the overlay** mode because 
- the free space left by the ribbon that lets you see the 3D view doesn't let you to interact with it, if you instead hide the ribbon you can interact with all the free space
- OP mentioned to add the tools that other overlay-ed panels have but already is there!
[image]

- last changes make the ribbon to not behave as previously: hivering over quick access buttons  activate the "move" cursor is shown which ibnterferes with normal functionning, quick acees buttons don't activate (you need to press 2 times fast), scroll no longer working, WB tab and ribbon

raindropsfromsky, 2024-12-25:
> Just tested it with those settings and it should work!

I don;t think your settings are the same. The canvas should look dark with my settings.

raindropsfromsky, 2024-12-25:
I tried again, and this time it worked, but partly.
[image]
As shown, the Ribbon UI has become transparent, but it has moved to one side.

hasecilu, 2024-12-25:
> I tried again, and this time it worked, but partly. [...]

Did you get problems trying to click/scroll on the ribbon?=

raindropsfromsky, 2024-12-25:
> Did you get problems trying to click/scroll on the ribbon?

No, but the ribbon and other panels gave me total grief by moving to unpredictable places. Cannot even move them back, as docking them is not so easy!

raindropsfromsky, 2024-12-25:
My ribbon has jumped to the left side, and I am unable to dock it to the top of the screen.
Even after unlocking the toolbars, I do not see its header bar anywhere, so I am unable to dock it!

[image]

raindropsfromsky, 2024-12-25:
> It is probably that your tree is very wide. You can stretch it near the top where the black bar becomes visible

No, I changed its width, and even moved the other panels. But I am unable to drag and dock the ribbon because I cannot see its header bar.

raindropsfromsky, 2024-12-25:
Update: I moved the other panels up. This resulted in buynching of all the panels at the top. FC showed a separate tab for sach panel. After that, I dragged the tab of the Ribbon and docked it across the top of the screen. (Phew!)

APEbbers, 2024-12-25:
Okay, I'm adding this as an option. Not as a default. 
And I place a warning with a help text next to the setting.
[image]

APEbbers, 2024-12-25:
@hasecilu  and @raindropsfromsky  Thanks for testing!

APEbbers, 2024-12-25:
@dgrabulosa I've added the origina Overlay function of FreeCAD as an option in the preference menu.
As you can see in the conversations, it can cause issues. That is why it is switched of by default.

If you have problems, place a file without extension named "OVERLAY_DISABLED" in the folder of the add-on. The Ribbon will be restored with the overlay function from this add-on.

kind regards

closed 2024-12-25 by APEbbers

dgrabulosa, 2024-12-26:
Ok, thanks for all effort put on these, I will test.