APEbbers/FreeCAD-Ribbon #217: When I launch a sketch in Part Design WB, FC switches to Sketcher WB, but the Ribbon UI remains blank (shows no toolbars)

raindropsfromsky, 2024-12-25:
When I launch a sketch in Part Design WB, FC switches to Sketcher WB, but the Ribbon UI remains blank (shows no toolbars)

Since this problem is persistent, I am raising a new bug.
(earlier it was reported as a part of another bug).

[image]

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * Beltrami 1.1.0
  * CommandPanel
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.34
  * frame 0.1.1
  * FrameForge 0.1.0
  * FreeCAD-Ribbon 1.6.0
  * freecad.gears 1.3.0
  * FreeGrid 2.1.1
  * OpenTheme 2024.9.1
  * parts_library
  * SearchBar 1.2.1
  * sheetmetal 0.6.13
  * Silk 0.1.5
  * symbols_library

APEbbers, 2024-12-25:
HI,

I cannot reproduce it at the moment, but I'll keep looking. Do you have set the sketcher wb to be autoloaded?
If not, can you test over a few days if it helps when it is set to autoload?

APEbbers, 2024-12-25:
Forget the previous message, the fix was not in the develop branch. It is in now.

closed 2024-12-25 by APEbbers