APEbbers/FreeCAD-Ribbon #237: Unclear notice about additional packages SearchBar Workbench

jace, 2025-01-08:
In FreeCAD 1.0 (macOS), installing from Addon Manager throws up this warning:

[image]

How do I install these dependencies? I tried from the CLI but that gave me another error:

```shell
$ /Applications/FreeCAD.app/Contents/Resources/bin/python -m pip install SearchBar Workbench
Collecting SearchBar
  Downloading SearchBar-0.6.tar.gz (10.0 kB)
ERROR: SearchBar from https://files.pythonhosted.org/packages/3f/92/d8c56b53d0e1a4df1ec4b9a5d9d0320367be4cd254fff7cc4cdaed6d06aa/SearchBar-0.6.tar.gz does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.
```

APEbbers, 2025-01-08:
Hi,

This indeed a confusion message. The SearchBar Workbench is actually an addon that you can install from the addon manager.
You can ignore this and continue with the installation.  When you have the searchbar installed it will be integrated in the ribbon
 
I'll create an issue with FreeCAD to change the message and behaviour during install.

Kind regards

APEbbers, 2025-01-08:
Hi @luzpaz,

Is there progress regarding the option to add an addon as an optional dependency in package.xml?
I'vecurrently  added the line according [https://github.com/APEbbers/FreeCAD-Ribbon/issues/56#issuecomment-2441661352] in my package.xml. I believe this issue is related to this.

Or is it better to  remove this line?

kind regards

APEbbers, 2025-01-08:
I think it is the ' at the end. I've updated the package.xml in the main branch with version 1.5.1.3 and with the develop branch.

luzpaz, 2025-01-08:
Ah, yes, I missed that. Nice catch!

jace, 2025-01-09:
FWIW, I do have the Searchbar Workbench installed, but I think I installed it later. And I'm so used to Python package names not having spaces that I read it as two separate items.

jace, 2025-01-09:
It's beachballing on a fresh install with this message in the report view:

[image]

APEbbers, 2025-01-09:
Hi,

Just checking, you have FreeCAD Ribbon version 1.6.0 Installed and FreeCAD's overlay not used?
In my test enviroment with MacOS, the ribbon is showing after updating to version 1.6.0.

Incase you have FreeCAD's overlay toggled, try opening a file from finder and see if the ribbon appears.

The message you see is just a warning, it shouldn't prevent the ribbon from loading. I have it to, but the ribbon shows

[image]

APEbbers, 2025-01-09:
Hi @jace,

The message for negative sizes is fixed with version v1.6.0.1. Gan you give it a try?

jace, 2025-01-10:
Should I Install manually? I used the Addon Manager last night.

APEbbers, 2025-01-10:
The addon manager is just fine. If it is easier you can use git from cmd but it shoudn't matter.

APEbbers, 2025-01-14:
Hi @jace,

Did get the ribbon working on your macOS?

kind regards

jace, 2025-01-15:
I had to refresh local cache in addon manager before it showed me the new version. It's now looking like this:

[image]

1. I'm using FreeCAD Classic theme, which defers to system light/dark mode. I guess Ribbon is assuming light mode here?
2. The window is  far wider than my monitor and doesn't allow resizing. Maybe I have too many workbenches? I think Ribbon is supposed to have its own overflow, so it's unclear what's happening here.
3. Two workbenches (MeshRemodel and Woodworking) cause FreeCAD to freeze without recovering. The console doesn't say much:

```
(qt.qpa.fonts) Populating font family aliases took 228 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost.
Activating Ribbon Bar...
SearchBox: Tools were loaded from the cache.
FreeCAD Ribbon: https://github.com/APEbbers/FreeCAD-Ribbon
2025-01-16 01:16:46.081 freecad[12985:29276747] +[IMKClient subclass]: chose IMKClient_Modern
2025-01-16 01:16:46.081 freecad[12985:29276747] +[IMKInputSession subclass]: chose IMKInputSession_Modern
An icon retrieved from data file for 'Toggle grid'
An icon retrieved from data file for 'Toggle snap'
An icon retrieved from data file for 'Configure rendering order'
Unknown command 'Std_MacroStopRecord'
(qt.qpa.window) Window position QRect(3047,1655 800x36) outside any known screen, using primary screen
[1]    12985 terminated  /Applications/FreeCAD.app/Contents/MacOS/FreeCAD
```

jace, 2025-01-15:
In a different place after restarting FreeCAD:

```
Oops. An unexpected error occurred:
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1216, in onWbActivated
    "color: " + StyleMapping.ReturnStyleItem("Border_Color") + ";"
    ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

TypeError: can only concatenate str (not "NoneType") to str
```

APEbbers, 2025-01-15:
Hi,

I've updated the Ribbon. to version 1.6.2.1.  Can you test if this resolve the issues?

FYI, the FreeCAD classic theme is replaced by FreeCAD Light. FreeCAD light has a lot of fixes in it. 
The ribbon works best with FreeCAD Light & Dark + OpenTheme Light & Dark.

jace, 2025-01-15:
I've switched to OpenDark. The errors have stopped and the window is now only slightly larger than my screen (see crop at left side). The icons are a mix of light and dark across workbenches:

[image]
[image]

I switched to FreeCAD Classic theme because it has cleaner borders than the Light and Dark themes and syncs with the system settings (mine is set to sunset-to-sunrise dark mode). I'll upgrade and report.

jace, 2025-01-15:
The workbench freezes stopped after I asked SearchBar Workspace to refresh its cache. Looks like that wasn't related to Ribbon.

jace, 2025-01-15:
Multiple errors after update and restart:

```
SearchBox: Tools were loaded from the cache.
FreeCAD Ribbon: https://github.com/APEbbers/FreeCAD-Ribbon
Uncaught exception:
   File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1216, in onWbActivated
    "color: " + StyleMapping.ReturnStyleItem("Border_Color") + ";"
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./StyleMapping.py", line 82, in ReturnStyleItem
    for key, value in StyleMapping_default["Stylesheets"]:
        ^^^^^^^^^^

ValueError: not enough values to unpack (expected 2, got 0)
Traceback (most recent call last):
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1216, in onWbActivated
    "color: " + StyleMapping.ReturnStyleItem("Border_Color") + ";"
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./StyleMapping.py", line 82, in ReturnStyleItem
    for key, value in StyleMapping_default["Stylesheets"]:
        ^^^^^^^^^^
ValueError: not enough values to unpack (expected 2, got 0)
CRITICAL:pyqtribbon_local.logger:Uncaught exception:
   File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1216, in onWbActivated
    "color: " + StyleMapping.ReturnStyleItem("Border_Color") + ";"
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./StyleMapping.py", line 82, in ReturnStyleItem
    for key, value in StyleMapping_default["Stylesheets"]:
        ^^^^^^^^^^

ValueError: not enough values to unpack (expected 2, got 0)
Traceback (most recent call last):
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1216, in onWbActivated
    "color: " + StyleMapping.ReturnStyleItem("Border_Color") + ";"
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./StyleMapping.py", line 82, in ReturnStyleItem
    for key, value in StyleMapping_default["Stylesheets"]:
        ^^^^^^^^^^
ValueError: not enough values to unpack (expected 2, got 0)
2025-01-16 01:59:36.692 freecad[16434:29333095] +[IMKClient subclass]: chose IMKClient_Modern
2025-01-16 01:59:36.692 freecad[16434:29333095] +[IMKInputSession subclass]: chose IMKInputSession_Modern
Uncaught exception:
   File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1216, in onWbActivated
    "color: " + StyleMapping.ReturnStyleItem("Border_Color") + ";"
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./StyleMapping.py", line 82, in ReturnStyleItem
    for key, value in StyleMapping_default["Stylesheets"]:
        ^^^^^^^^^^

ValueError: not enough values to unpack (expected 2, got 0)
Traceback (most recent call last):
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1216, in onWbActivated
    "color: " + StyleMapping.ReturnStyleItem("Border_Color") + ";"
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./StyleMapping.py", line 82, in ReturnStyleItem
    for key, value in StyleMapping_default["Stylesheets"]:
        ^^^^^^^^^^
ValueError: not enough values to unpack (expected 2, got 0)
CRITICAL:pyqtribbon_local.logger:Uncaught exception:
   File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1216, in onWbActivated
    "color: " + StyleMapping.ReturnStyleItem("Border_Color") + ";"
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./StyleMapping.py", line 82, in ReturnStyleItem
    for key, value in StyleMapping_default["Stylesheets"]:
        ^^^^^^^^^^

ValueError: not enough values to unpack (expected 2, got 0)
Traceback (most recent call last):
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1216, in onWbActivated
    "color: " + StyleMapping.ReturnStyleItem("Border_Color") + ";"
[... 153 lines cut ...]
    for key, value in StyleMapping_default["Stylesheets"]:
        ^^^^^^^^^^
ValueError: not enough values to unpack (expected 2, got 0)
CRITICAL:pyqtribbon_local.logger:Uncaught exception:
   File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1216, in onWbActivated
    "color: " + StyleMapping.ReturnStyleItem("Border_Color") + ";"
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./StyleMapping.py", line 82, in ReturnStyleItem
    for key, value in StyleMapping_default["Stylesheets"]:
        ^^^^^^^^^^

ValueError: not enough values to unpack (expected 2, got 0)
Traceback (most recent call last):
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./FCBinding.py", line 1216, in onWbActivated
    "color: " + StyleMapping.ReturnStyleItem("Border_Color") + ";"
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jace/Library/Application Support/FreeCAD/Mod/FreeCAD-Ribbon/./StyleMapping.py", line 82, in ReturnStyleItem
    for key, value in StyleMapping_default["Stylesheets"]:
        ^^^^^^^^^^
ValueError: not enough values to unpack (expected 2, got 0)
```

FreeCAD has opened with only the search toolbar (and a small window size!)

[image]

APEbbers, 2025-01-15:
> I had to refresh local cache in addon manager before it showed me the new version. It's now looking like this:
> 
> <img alt="Screenshot 2025-01-16 at 01 12 38" width="1766" src="https://private-user-images.githubusercontent.com/26695/403564327-adc93b28-0fbd-42e9-a47d-7a33f5acc006.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY5NzE1MTYsIm5iZiI6MTczNjk3MTIxNiwicGF0aCI6Ii8yNjY5NS80MDM1NjQzMjctYWRjOTNiMjgtMGZiZC00MmU5LWE0N2QtN2EzM2Y1YWNjMDA2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTE1VDIwMDAxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMxYmNmZDBjOGU2NzcwOTEzMDk0YmMwODEwYTQwZTQyODBlYWQxNGQwYjdiYjZkYWFkOTRkMjI1OWEzMjUxOWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.svc79gMEg3YJiVgdmanQP6q6VSr9lQVUQmpYiLrBk24">
> 1. I'm using FreeCAD Classic theme, which defers to system light/dark mode. I guess Ribbon is assuming light mode here?
> 2. The window is  far wider than my monitor and doesn't allow resizing. Maybe I have too many workbenches? I think Ribbon is supposed to have its own overflow, so it's unclear what's happening here.
> 3. Two workbenches (MeshRemodel and Woodworking) cause FreeCAD to freeze without recovering. The console doesn't say much:
> 
> ```
> (qt.qpa.fonts) Populating font family aliases took 228 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost.
> Activating Ribbon Bar...
> SearchBox: Tools were loaded from the cache.
> FreeCAD Ribbon: https://github.com/APEbbers/FreeCAD-Ribbon
> 2025-01-16 01:16:46.081 freecad[12985:29276747] +[IMKClient subclass]: chose IMKClient_Modern
> 2025-01-16 01:16:46.081 freecad[12985:29276747] +[IMKInputSession subclass]: chose IMKInputSession_Modern
> An icon retrieved from data file for 'Toggle grid'
> An icon retrieved from data file for 'Toggle snap'
> An icon retrieved from data file for 'Configure rendering order'
> Unknown command 'Std_MacroStopRecord'
> (qt.qpa.window) Window position QRect(3047,1655 800x36) outside any known screen, using primary screen
> [1]    12985 terminated  /Applications/FreeCAD.app/Contents/MacOS/FreeCAD
> ```

2. It seems that FreeCAD itself is very wide. On your screenshot "FreeCAD 1.0.0" is on the right while in my screenshot earlier, this is in the middle.
3. I've tested the meshremodel and woodworking workbenches and they working on my side.  But the issue can be the last line in your second picture. Can you start freecad minimized or if you drag it to the right, you can maybe stretch it from the left and see if that helps?

jace, 2025-01-15:
The workbench issue went away after refreshing SearchBar's cache. However, Ribbon 1.6.2 is now showing these errors on all themes and failing to render toolbars, and FreeCAD is segfaulting on exit. I'll nuke `system.cfg`, `user.cfg` and searchbar cache and try again (Searchbar caused segfaults for me previously, without Ribbon installed).

APEbbers, 2025-01-15:
I was too fast. with verion 16.2.2, the issue should be fixed.
[image]

jace, 2025-01-15:
No more errors or segfaults on this version, but the window is too wide again and the icons have bright backgrounds in the ProDark theme (I haven't yet nuked config). This is the smallest the window can be resized to:

[image]

APEbbers, 2025-01-15:
Actually you do see now the entire width of FreeCAD. On my system, I can stretch the window, So it is strange that on your system it doesn;t work. Which version of MacOS are you using?

As for the background. Try to switch to FreeCAD Dark or OpenDark. These are supported. Unsuported themes get the default background. which is white.

[image]

jace, 2025-01-15:
I changed theme from ProDark to OpenDark and the icons are re-rendering in dark mode when I hover over them, so it looks like Ribbon missed a refresh on theme change. Now I have mixed colour icons (and window still too wide):

[image]

(This is after switching back to ProDark. Light background icons are ProDark, dark background are OpenDark.)

I'm on macOS 15.1.1 with a 4K monitor scaled to 3008x1692.

jace, 2025-01-15:
FreeCAD version info (installed from Homebrew, no update available):

OS: macOS 15.1.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: ProDark.qss/ProDark/Fusion
Installed mods: 
  * MeshRemodel 1.10.35
  * SearchBar 1.3.3
  * Reporting
  * DynamicData 2.70.0
  * FreeCAD-Ribbon 1.6.2.2
  * FreeCAD-themes 2025.1.7
  * freecad.gears 1.3.0
  * woodworking 0.21.2.33771.0.23.1
  * OpenTheme 2024.9.1
  * FreeGrid 2.1.1
  * Defeaturing 1.2.3
  * pcb 6.2023.1
  * fasteners 0.5.35
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.3
  * MakerWorkbench 1.0.1
  * Curves 0.6.53

APEbbers, 2025-01-15:
Just checking, after changin the theme, do you restart FreeCAD? it might help

jace, 2025-01-15:
I switched to OpenDark and restarted and now the window is resizable and fitting on screen and the icons are correct! Some workbenches still show an extra toolbar on first use, but switching out and back fixes it.

APEbbers, 2025-01-15:
I've not used homebrew on my installation because its installed on virtualbox and I don't have xcode.
I've downloaded FreeCAD from the website. You get an FreeCAD conda version. can you try if that changes the behavior of the window size?

jace, 2025-01-15:
Everything is working now.

Just one issue left: the DynamicData workbench adds a Commands toolbar to other workbenches that renders a few seconds later, outside the ribbon. Here it is in the Gears workbench:

[image]

jace, 2025-01-15:
> As for the background. Try to switch to FreeCAD Dark or OpenDark. These are supported. Unsuported themes get the default background. which is white.

Can't Ribbon use theme colours here?

APEbbers, 2025-01-15:
> Everything is working now.
> 
> Just one issue left: the DynamicData workbench adds a Commands toolbar to other workbenches that renders a few seconds later, outside the ribbon. Here it is in the Gears workbench:
> 
> <img alt="Screenshot 2025-01-16 at 02 39 05" width="1478" src="https://private-user-images.githubusercontent.com/26695/403585385-78503169-9137-4a25-9732-0bf8ea289329.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY5NzU3MTYsIm5iZiI6MTczNjk3NTQxNiwicGF0aCI6Ii8yNjY5NS80MDM1ODUzODUtNzg1MDMxNjktOTEzNy00YTI1LTk3MzItMGJmOGVhMjg5MzI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTE1VDIxMTAxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU1ODYwMTdjYWQzZjg4OTA1NmFmMDIzZDEyYzgzYjIyYzBmZWVjMzUxNWVhY2U4YTUxN2U0YjA5YmIwNWY4MmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Aa2M7qbC2lAbwPTlHQLpWh25BZubeisEn4sRc2OsrDo">

When you click on a tab, the toolbar should dissapear again. I believe the DynamicDataWB has an setting to disable this toolbar.
[image]
 Uncheck the top setting in the picture above:

APEbbers, 2025-01-15:
> > As for the background. Try to switch to FreeCAD Dark or OpenDark. These are supported. Unsuported themes get the default background. which is white.
> 
> Can't Ribbon use theme colours here?

I'll look into that, but it might take me a bit longer

jace, 2025-01-16:
> I'll look into that, but it might take me a bit longer. problem here is that the default background color of FreeCAD itself is also white. The themes are created with stylesheets and there are setup differently for each. This makes it difficult to get the correct color. That's why only FreeCAD light and dark + Opentheme is supported currently. These are properly integrated.

All right. Thanks for all the work that went into this add-on! Looks like I'm all set now. Minor quibbles:

1. The confusing message about Python dependencies (original report here) appeared again when I did the last update, even though SearchBar Workbench was already installed.
2. Ribbon's Menu has Ribbon-specific items (customize, help, about) that do not appear in the macOS menu. I'm used to hitting Cmd+? to use the native search for menu items.
3. Maybe support all the first-party themes in the [FreeCAD-themes add-on](https://github.com/FreeCAD/FreeCAD-themes)? That'll be less confusing for the next person checking out themes.
4. The Tools toolbar appears in all workbenches, but Ribbon puts it last so the on-screen location keeps changing. Maybe put it first (alongside Structure) for consistency? (I did this manually from Ribbon Layout settings.)

luzpaz, 2025-01-16:
Great teamwork here. Thanks for all the work!

APEbbers, 2025-01-16:
Hi,

I've added points 2 to 4 to my roadmap for the next release of the Ribbon.  For the first point, i've created an issue for FreeCAD about it. For now I'm closing this issue

Indeed thank you for bringing these issues up and the help with testing.

Kind regards

closed 2025-01-16 by APEbbers

luzpaz, 2025-01-19:
> i've created an issue for FreeCAD about it

@APEbbers can you link to it from this ticket ?

jace, 2025-01-25:
Update: As of 1.6.2.5, the non-resizable window problem is back. FreeCAD is now as wide as the Ribbon needs to display all workbenches. I haven't used FreeCAD in the past week. This is just after re-open and update.

[image]

APEbbers, 2025-01-26:
Hi, that's strange, nothing has changed in the code itself. Only the package.xml file to fix the message during install and some additional stylesheet settings for support of other themes.

Are you sure that you are opening the FreeCAD downloaded from the freecad website and not the FreeCAD from Homebrew?
(just checking?

jace, 2025-01-28:
I'm using FreeCAD 1.0 from Homebrew, same as before. This happened after I tried switching themes to check if Ribbon supported them now, and since then it's remained persistent no matter what theme I use.

I think it's the same trigger as earlier, something within FreeCAD, but I don't know my way around the code yet. Maybe if I reset settings to last known-good it'll sort itself out.

APEbbers, 2025-01-28:
Last time you tried the version from the FreeCAD website instead of homebrew. Then you had no issues. I think there is something wrong with the Homebrew version of FreeCAD.  Maybe it is missing a library or something

[image]

luzpaz, 2025-01-28:
Maybe worth opening a ticket on https://github.com/FreeCAD/homebrew-freecad ?

APEbbers, 2025-01-28:
@jace: Can you make a ticket on [ https://github.com/FreeCAD/homebrew-freecad ](url)? I'm not able to install homebrew because I don't have xcode and cannot test anything regarding homebrew compilations and installations.

kind regards

jace, 2025-01-29:
Sorry if I wasn't clear earlier. I was always using the Homebrew version. [Homebrew's formula for FreeCAD](https://github.com/Homebrew/homebrew-cask/blob/fa21615a8d24e3579de7bb77d09eb66b51fe55df/Casks/f/freecad.rb) downloads it from their official releases on GitHub, so it's the same download as on the website. I'll try a nightly build too in case this has been fixed in a pre-release.

APEbbers, 2025-01-30:
Hi,

I finally could reproduce your error. It occurs when you select "FreeCAD Classic". Then the Ribbon get's too wide. 
Best to avoid this theme for now. If you want to use "FreeCAD classic," use "FreeCAD light" instead.
FreeCAD Light is the successor of FreeCAD Classic with a lot of bugfixes.

I'm currently working on release 1.7 of the FreeCAD Ribbon. Here the focus is on UI optimalization for MacOS.
I'll have a look if I can fix the issue for the FreeCAD Classic theme in the next release.

In my develop branch, I've added the menus and extra support for themes. if you like you can switch to the develop branch already to test it.

Kind regards

jace, 2025-01-31:
Aha! Is the problem sticky even after switching away from the Classic theme? I think I clicked through them one by one before settling on OpenDark and got stuck with the wide window despite restarts.

It seems like an inverted dependency somewhere: instead of Ribbon adding scroll buttons based on the window width, the window is getting min-width from Ribbon's tab bar.

If it's sticky it may be going into one of the CFG files. Does it have ordered load so reordering can fix the problem?

Wildly speculating here, sorry!

luzpaz, 2025-01-31:
Nice catch @APEbbers