APEbbers/FreeCAD-Ribbon #256: Feature request: Integrate the Ribbon-related menu options in the menu

raindropsfromsky, 2025-01-31:
Currently the Ribbon-related menu options are given undue prominance.
[image]

These menu items pertain to the Mirror WB, which is one of the many WBs of FreeCAD.
If all WBs start pushing their options in the main menu, the menu will become bloated and cluttered; and impossible to navigate!

For example, the main menu already has two entries related to Sketcher and Part Design WBs.
They also need to be cleaned up.

Note that no other WBs have intruded in the menu system.
- If anyone goes through the menu, he will not realize that all functionalities of FreeCAD are accessible only through WBs.
- When we select any WB, the menu does not change at all to provide those functionalities thorugh the menu system.

So, only three WBs (Sketch, Part Design, Mirror) have broken that rule.

At present, the menu structure is broken in multiple ways:
1. The **Help** option repeats twice (once for the FreeCAD itself, and then for the Mirror WB.)
2. The **About** menu is always is always tucked away as "Help > About".
    But here, the "About" item is promoted by two levels, to become at par with the Help of FreeCAD!
3. The **What's new** entry is usually tucked away as **Help > About > What's new** menu.
    Instead, here it is promoted by four levels!

Instead, all Ribbon-related menu should be placed in appropriate place in the menu tree.
Further, instead of placing multiple ribbon-related menu items at the top level, there should be a single entry in the main menu (let's call it **Ribbon**). Demote all Ribbon menus under this entry.

The **Ribbon** entry should be placed above the **Help** menu. 
[image]

(In any software, **Help** is always the last entry in the menu, so that the user can access it easily.)

raindropsfromsky, 2025-01-31:
On second thoughts, I realized that the Ribbon "workbench" actually is not a real workbench like Sketcher or Part Design. 
It is a interface style, which is running at all times, along with the selected workbench.

Therefore, any controls to control this style of interface should appear in a distinct area, so that the user does not confuse them with the controls for the currently selected workbench.

So, we can have a set of three small icons in the far left side of the ribbon itself.

For example, the settings, overlay and help menus for the ribbon interface can be shown as-
[image]

raindropsfromsky, 2025-01-31:
Even version 1 is already fantastic! 👍🏽👍🏽

But so far as the menu system is concerned, there need not be a user intervention.
It only needs a little re-organization out of the box.

raindropsfromsky, 2025-02-01:
I changed to Development branch, and it installed v 1.6.2.6.
I still think this is not the right version, because it does not have the changes I was hoping to see.

Please see if I have selected the correct branch:
[image]

All the Ribbon-related menus are still at the top level (not tucked under a **Menu > Ribbon...** node.)
[image]

I am also unable to see the 3-icon toolbar that you have shown at far right of the screen.

IMO it should be placed at the far left of the Ribbon UI, as shown below:
[image]

The advantage is that even when the user selects any WB, this section remains in the same place (persistent).

APEbbers, 2025-02-01:
Hi,

No you are still on an older version. Perhapse you have to update again after switching to the develop branch.
The branch is correct by the way.

Those 3 buttons you show on the left, are placed on the top right toolbar. That remains also the same when the user changes the wb.
I haven't seen a CAD program that has those kind of buttons on the left of the ribbon (SolidWorks, SolidEdge, NX, etc.). I believe it is best the leave the ribbon itself soley for the cad commands.

kind regards

raindropsfromsky, 2025-02-01:
Strange!

when I took the screenshot to explain the situation, I had already done the following:
1. switched to the Development branch, and 
2. pressed the **Check for Update** button.
3. Installed the latest update in Ribbon
4. Restarted the FC

So the **change branch** command was just a dummy operation for me when took the screenshot.
That's why the branch is already 1.6.2.6.

Yet I am unable to see the changes you showed.

BTW, when I change the branch, it does show an error message:
```
Some changes are not committed.
Do you still want to switch?
```
Could that be the reason?

raindropsfromsky, 2025-02-01:
I see this:

[image]

raindropsfromsky, 2025-02-01:
Update: I switched to **pre-release 1.7** branch.

Now I see the changes:

[image]

APEbbers, 2025-02-01:
Great! one issue remaining is the color of tooltips. other than that, version v1.7.0 is ready to release.

raindropsfromsky, 2025-02-01:
This is fantastic development, indeed!

Just one small niggle: When we click on the [image] icons, they do not respond. 

We have to click on the drop-down arrows to see the options.
The problem is that the arrow is a tiny target area.

Can you make the whole button as the target for clicking?

raindropsfromsky, 2025-02-01:
DRUMROLLS!

Now the UI looks very mature, and fantastic!
The Ribbon transforms FC into a modern-looking app, at par with the best out there!
Surely many new users will be attracted to FC if the Ribbon WB gets enough publicity through YouTube channels.

--------
Just a small niggle:

In the menus, the word "FreeCAD" is redundant.
We are already in FreeCAD, so the user cannot mistake it for any other app! 

[image]
[image]

It is desirable to shorten the label.

If at all, we can opt for "Ribbon UI" (if just "Ribbon" looks a bit hazy).

APEbbers, 2025-02-01:
Thanks! 
Aha yes I was dubbing about that. The project is now called FreeCAD Ribbon, that's why I put it in like this. But Ribbon UI is indeed better. I'll have to change to tooltip of the application button as well.

[image]

Something like: "Ribbon application menu..."

raindropsfromsky, 2025-02-01:
I found one inconsistency in the UI management:

The **FreeeCAD preferences** UI allows the user to choose the type of WB sector (, and its display style (icons/text/icon+text)

[image]

This is not applicable when the Ribbon WB is installed.
This conflict should be resolved.

raindropsfromsky, 2025-02-01:
> Something like: "Ribbon application menu..."

No, the word "application" will confuse the user.
Just Ribbon is good enough, and intuitive.

For comparison, MS Office has switched to ribbon long ago.
Now LibreOffice also offers the traditional toolbars and a ribbon interface.
So ribbons are well-understood.

raindropsfromsky, 2025-02-01:
>  The project is now called FreeCAD Ribbon

The project should be called "Ribbon UI" (the word "FreeCAD" should be dropped.)

Also, it should not be labeled as a WB, because the ribbon is just an interface style.
It organizes the toolbars, sections, and buttons of all WBs as a ribbon, but it is not a WB by itself.

BTW, can the Ribbon be activated in a vanilla FC by default? :)

APEbbers, 2025-02-01:
> I found one inconsistency in the UI management:
> 
> The **FreeeCAD preferences** UI allows the user to choose the type of WB selector (tab bar/combobox), and its display style (icons/text/icon+text).
> 
> [image]
> 
> However, both these oiptions are not applicable when the Ribbon WB is installed. This confusion should be resolved.

This is not possible because the Ribbon is now an add-on instead of an core part of FreeCAD. For me to remove this would mean to change the core of FreeCAD. So unfortunally this is as is for the moment

APEbbers, 2025-02-01:
> The project should be called "Ribbon UI" (the word "FreeCAD" should be dropped.)
>
> Also, it should not be labeled as a WB, because the ribbon is just an interface style. It organizes the toolbars, [...]

I know they want ot intergrate or use parts of the ribbon. For that i have to change the license. I have planned to do that after I have version 2.0 ready. Then they can intergrate or takeover the ribbon if they want

APEbbers, 2025-02-01:
> No, the word "application" will confuse the user. Just "Ribbon" is good enough, and intuitive.
>
> For comparison, MS Office has switched to ribbon long ago. Now LibreOffice also offers the traditional [...]

> The project should be called "Ribbon UI" (the word "FreeCAD" should be dropped.)
>
> Also, it should not be labeled as a WB, because the ribbon is just an interface style. It organizes the toolbars, [...]

Mayby I just show the shortcut. "menu" on the button is clear enough i think.

raindropsfromsky, 2025-02-01:
The "About" link is missing.
IMHO it can be added in the Ribbon's Help webpage itself.
(That way, FreeCAD's menu system remains uncluttered.)

raindropsfromsky, 2025-02-01:
> This is not possible because the Ribbon is now an add-on instead of an core part of FreeCAD. For me to remove this would mean to change the core of FreeCAD. So unfortunally this is as is for the [...]

Ah well!

Anyhow once they adopt the Ribbon UI, they will be able to change the UI and make it seamless.
So not a biggie!

APEbbers, 2025-02-01:
> The "About" link is missing. IMHO it can be added in the Ribbon's Help webpage itself. (That way, FreeCAD's menu system remains uncluttered.)

The are in the help menu? Or do you mean something else. (They both work from my side)

[image]

raindropsfromsky, 2025-02-02:
oops!

I missed observing that the **About...** is not a single menu option any more! 👍

closed 2025-02-07 by APEbbers