APEbbers/FreeCAD-Ribbon #257: Feature request: Provide basic guidance about how to use each screen of the customization

raindropsfromsky, 2025-01-31:
Ribbon WB has seen a huge change over the last few weeks.
It transforms FreeCAD into a modern-looking app, with a much better access to all the tools and customization to the user's heart's content.

However, this flexibility comes at a price: The customization interface has become complex.
Further, some parts of the interface use unconventional controls and layouts.
That's why it is best to guide the user.

Two examples are given below:

---------

**Example-1: How the columns work**
This screen allows two different options for customization:
1. He can set the icon size for whole workbenches. He can select multiple WBs at a time.
2. He can set the icon size for specific toolbars of specific WBs. He can select multiple sections from any of the WBs at a time.

The nuanced working as explained above is not clear from the screen layout.
At first glance, it looks like the user has to do something in the left list, and that will change the right list.
But here, both columns are designed to act independently.

Ideally, the two lists should not be presented at once. 
Let the user select a radio button, and then present only one list for that option.

--------
 
**Example-2: How the selection works in each list**
The selection is "sticky", which is unusual. 
The user has to just click to toggle the selection state of a given item.
When he clicks on any item, the selection state of the other items does not change.

Normally, we do not encounter this selection method at all.
We normally select items by pressing CTRL or SHFT and clicking.
Or we draw a rectangle to select multiple items.

------------

**How to provide tips to the user:**
This can be done in two ways, as shown below:
[image]

1. Provide extended tooltips on the buttons
2. Provide a short text field that explains what to do.

APEbbers, 2025-01-31:
Hi,

Thank you for your suggestions. For guidens a manual is present in my wiki page. (click on the help button in the layout dialog or on "Ribbon help" in the application menu)

For version 2 of the FreeCAD Ribbon, I'll introduce drag and drop functionalities and context menus. The plan is to simplify this dialog and move most of its current functions to context menus. I'll have a look if I can implement example 2 in version 2 of the ribbon.

Regarding suggestion 1: tab names and guiding texts are quite self-explanitory in my opinion. I don't believe that tooltips will add any new information.

kind regards

raindropsfromsky, 2025-01-31:
I would like to contribute to the help documentation.
Plerase let me know!

APEbbers, 2025-01-31:
HI,

Thanks! I've set the wiki open for editing. The link is: https://github.com/APEbbers/FreeCAD-Ribbon.wiki.git.
Please don't change chapters 06 and 07. These are my change logs and roadmaps which I'd like to manage myself only.

kind regards

raindropsfromsky, 2025-01-31:
The link is dead (I get 404 error)

APEbbers, 2025-01-31:
it's the link to clone it with git. Here is the link for in your browser: https://github.com/APEbbers/FreeCAD-Ribbon/wiki