APEbbers/FreeCAD-Ribbon #259: Bug: The button for the workbench is invisible in the tab bar

raindropsfromsky, 2025-01-31:
I am using dark theme with FC.
The button for one of the workbenches has a very low contrast. Practically invisible.

[image]

APEbbers, 2025-01-31:
Hi,

I'll have a look if I can somehow invert the colors. But this actually more an issue for the Dodo workbench rather than the Ribbon.
The Dodo workbench should provide a different icon for dark themes.

kind regards

APEbbers, 2025-01-31:
I'll have a look first. If I can create some function, that would help also with future wb's. 
But each wb has its own github. You should raise wb related issues there otherwise they won't be picked up.

APEbbers, 2025-01-31:
If I look in the add-on manager it seems that the DodO workbench is replaced with the "quetzal" workbench. That icon has no problems:

[image]

I'm closing this issue for now

closed 2025-01-31 by APEbbers

raindropsfromsky, 2025-02-01:
oops! I did not realize that the Dodo workbench has already gone the way of Dodo (extinction)! 😄