APEbbers/FreeCAD-Ribbon #261: Feature request: Please provide extended tips for the Workbench bar

raindropsfromsky, 2025-01-31:
Cuuerently, the workbench bar does not have even simple tooltips.
[image]

I have selected "icons only" mode for the WB bar, as it is very space-efficient.
As a newbie, I am using only 2-3 workbenches.
However, when I need to select one of the other workbenches, I would like to know their names and pourpose on hover, through extended tooltips.

The extended tooltips should show the name of the workbench, followed by its main uses.

APEbbers, 2025-01-31:
Hi,

I'll add at least a simple tooltip. I'm not planning to write my own tooltips for each workbench. These should be provided by the workbenches themselfes. If there are extended tooltips available, I'll add these.

kind regards,

raindropsfromsky, 2025-01-31:
Thanks!

Should I raise this feature request with FC? 

IINW this should be included in their design guidelines to the WB developers, so all WBs comply with this interface requirement.

APEbbers, 2025-01-31:
I believe there is already some development for extended tooltips. i found this thread on discord: https://discord.com/channels/870877411049357352/1322642504998457456

raindropsfromsky, 2025-01-31:
Actually, all the toolbar buttons in all the workbenches are already producing the extended tooltips.
Only the tab bar (Workbench selector) does not have tooltips (either simple or extended).

That's why I thought that these tooltips are produced at your sole discretion, and they are not coming from the respective WBs.

BTW I can create a short text for all workbenches that can be used as extended tooltips.
Since new WBs are added rarely, we will need updates for these tooltips very rarely.

APEbbers, 2025-01-31:
No need. I found out that some workbenches do have a tooltip and an extra description. 
In my develop branch, I've added the tooltips.

For some workbenches you'll get an extended tooltip and when no extra description is available, you just get a short tooltip.
This way, all the information is retrieved from the workbench itself including any available translation.

[image]

[image]

The tooltips will become available with the next release in the main branch

closed 2025-02-07 by APEbbers