APEbbers/FreeCAD-Ribbon #264: Suggestion: There are two nearly-duplicate panels in Assembly WB. One of them should be removed.

raindropsfromsky, 2025-02-01:
The Assembly WB has a duplication:

[image]

This should be cleaned up for v 1.7

APEbbers, 2025-02-01:
That is probally an left-over from previous version.

Try to change this settings under preferences and see if that helps:

[image]

It should add the other variant of the view panel to the excluded panel list

raindropsfromsky, 2025-02-01:
My view was already set to Ribbon.
But the "User standards Tools panel" checkbox was unselected. I selected it.

But the duplicate entries are not removed.
In fact, I noticed that there are TWO duplicate pairs!

[image]

raindropsfromsky, 2025-02-01:
I have a different UI, as shown below.

[image]

Perhaps that's because I am using the latest weekly FC.

I did see two entries, which I delected and restarted FC.

Now I am seeing the following panels in Assembly WB:
Structure
Assembly
Assembly Joints
Views - Ribbon newPanel
Tools newPanel

While the duplicates are removed, the last two panels have strange labels!

APEbbers, 2025-02-01:
Hi,

Those labels arre on purpose. This way you can see that they are created with the "new panel" function. You can delete them there as well. The suffix "_newPanel" won't appear on the ribbon, but the code depends on them. 

The reason that you saw duplicates, is because as from V1.5.0, these panels are created differently as before. (with the new Panel function) I don't believe the weekly build is the issue. I'm also working on the developers version of FC.

kind regards

closed 2025-02-01 by APEbbers