APEbbers/FreeCAD-Ribbon #265: BUG: The "Ribbons Layout Screen" remains on top of other apps. It should be on top of FC window only

raindropsfromsky, 2025-02-01:
The dialog titled "Ribbon design" remains on the top of all applications running on the computer.
This should be changed to let it stay on top of FC window only.

[image]

raindropsfromsky, 2025-02-02:
After launching the **Ribbon design** dialog, if I switch to other app (using ALT+TAB), the dialog is no longer on top of that other app. 👍 

However, if I switch back to FC, the **Ribbon design** dialog is no longer there! 
In other words, the dialog vanishes if I switch back to FC.

**Desired:**
When I switch back to FC, the dialog should still be there on top of FC window.

APEbbers, 2025-02-04:
I've tried to find a solution in the documentation. Unfortunally, it is either always on top, or dissapearing when switching to another app. So I've decided to reverted back to always on top. You can always minimize the dialog itself manually.

kind regards

closed 2025-02-04 by APEbbers

raindropsfromsky, 2025-02-05:
oh no!

In fact, the behavior you set earlier was much better!

I see it this way: When we are in the middle of changing settings for an app, we are not likely to switch over to some other app. (That is distracted behavior!) So if we do switch over, it is ok if we have to start all over in FC. On the other hand, if the child window is on top of all windows, it is very irritating.

Could you please revert to the earlier modification?