APEbbers/FreeCAD-Ribbon #271: BUG: Some of the WB buttons bleed color on the right edge

raindropsfromsky, 2025-02-02:
I have set WB button size to 30.
While experimenting with the color settings, I discovered by chance that some of the buttons bleed color on the right edge.

[image]

The default color is a faint blue, in which this problem is not observable.

APEbbers, 2025-02-02:
Hi,

I couldn't repoduced, but I did changed something because I had a idea what it might be. Can you test if it is resolved?

raindropsfromsky, 2025-02-02:
I updated Ribbon, but the problem is still there.

I have a FHD monitor, set to 115% custom scale.
In the past, some apps have had display problems because of this.
Probably that is true for FC also?

APEbbers, 2025-02-02:
Hi, 

yes that might be the issue. Unfortunally Qt on which FreeCAD is based has also issues with this. It is a known bug.
The reason that it's only some wb tabs show it, is because i did a trick to make the tabs icon only.

The PyQtRibbon package on which the ribbon is build, requires text in the tab. otherwise it cannot determine tab indexes, etc. and will crash.
What I've done is make the text transparant and then changed the tab width via CSS stylesheets. This way, the text is still present, but that tab appears to be icon only.
What you see is the beginning of the text, which has the same color as its background. And there lays the issue with DPI scaling. The background is the tab but due to scaling, the text falls outside the tab partly. Even microsoft itself has still issues with DPI scaling and dialogs. (WIN10 mostly)

I',m unable to compensate for the scaling. I will keep it in mind and If have an idea, I'll come back to you.

kind regards

raindropsfromsky, 2025-02-02:
Maybe insert a space in the beginning of the dummy text, so that it does not fall in the boundary of the icon?

closed 2025-02-07 by APEbbers

raindropsfromsky, 2025-02-09:
In the latest development version, we can see some part of the label text now.
Please reopen the issue.

[image]

APEbbers, 2025-02-09:
Can you check if setting scaling to 100% solves the issue? 
Also, your on windows right? how do you set scaling to 115%? I've only these:
[image]

I've checked it on Windows, Linux and macOS with tabbar size set to 36, and it shows correctly. Also when I change the scaling.
It is a bit strange why it does occur only on your system. 

I'm not convinced that is a Ribbon issue, but perhaps it is specific for your setup

raindropsfromsky, 2025-02-09:
> how do you set scaling to 115%? I've only these:
https://chatgpt.com/share/67a8aff4-bc0c-8003-9e23-c7e6cfb9810c

I reset my scale to 100%, and it looks fine.
But it makes all fonts on my laptop look too tiny.
It also reduces the icons in my task bar too tiny.
On the other hand, 125% makes them look too large.
115% is the sweet spot for me.

raindropsfromsky, 2025-02-09:
At 125%, the color bleeding is there as I reported initially.
But at least the text does not appear inside the button border.

APEbbers, 2025-02-09:
I've set my system also to 115%. Still no issue on my side. even with custom colors enabled.
I don't now what is going on. Do you have any other setting related to your display, that might cause this behavior?

raindropsfromsky, 2025-02-10:
> Do you have any other setting related to your display, that might cause this behavior?
Is there a way to export all the settings so that you can replicate my settings and see what is going on?

raindropsfromsky, 2025-02-10:
When I reinstalled FC yesterday, I completed its set up, including enabling change of branch in AM.
Now I am getting text in the icons again.

[image]

I think some settings in the Preferences may be clashing with Ribbon settings.

[image]

[image]

APEbbers, 2025-02-10:
Hi,

can you send me your user.cfg and system.cfg files? I'll try to see any difference between settings.
[image]

kind regards

raindropsfromsky, 2025-02-10:
Do they have any user-sensitive info that I should delete first?