APEbbers/FreeCAD-Ribbon #272: Feature request: Where a button has a drop-down arrow to provide multiple options, please allow the user to click on the button itself to see those options

raindropsfromsky, 2025-02-02:
Currently, if a button has a drop-down arrow to provide multiple options, the user has to click on the arrow to see those options.
If he clicks on the main icon, there is no response.
The problem is that the button presents a very tiny target area (especially when the toolbar buttons are made small (e.g. 26 pix).

[image]

The solution is to allow the user to click on the button itself to see those options.
The button (including the drop-down arrow) presents a much larger target for clicking, so it will be very comfortable for the user.

[image]

--------------

I know that there is another implementation, in which the drop-down arrow selects one of the modes.
Then if the user clicks on the main button, the button works in the selected mode.
Thus, the button will continue to work in that mode till the user changes the mode by clicking on the drop-down arrow and selecting any other mode.

But in Ribbon, this method is not adopted.
So each time the user has to produce the options list and then select one option.

Therefore, the design is amenable to the idea that the entire button be used as clicking target to trigger the options list.

APEbbers, 2025-02-02:
Hi,

This is by design. The major CAD programs have the same behaviour as the ribbon. (SolidWorks, SolidEdge, etc.)
When you have a model loaded, the buttons will be functional and when you click on the main icon, that command is executed.
When clicking on the dropdown, you can select the other commands.

In your screenshots, the buttons are greyed out. This means that you don't have a model loaded or that the command is not applicable (often you have to select something first).
[image]

kind regards,

closed 2025-02-02 by APEbbers

raindropsfromsky, 2025-02-02:
I took a random screenshot just to show the icon area and the arrow.

I have picked another animated screencapture where the button is not gray, which means that right from the first click, it should do something. Further, if I change its option from its menu, then the new option should remain persistent, and from that moment onward, if I click on the button area, it should execute the latest selected option. 

Well, as the attached file shows, this does not happen: The icon of the button does not change when I select a new option.
So whenever I click on the button, it will execute only one of the six options.
To execute all the other five options, I have to select from the menu.

[image]

APEbbers, 2025-02-02:
It just works:

https://github.com/user-attachments/assets/44e43aaf-cd97-4cde-9b15-2adb4bda2558

raindropsfromsky, 2025-02-02:
Oh no doubt it works, but every time we have to select a particular mode.
You cannot click the main button to repeat the last command, because you can't tell what it was (because the icon has not changed).

In Blender, the icon changes, and that's a very good thing.
All the states of a button are recognizable easily, so the user would never face confusion as to what that button does.