APEbbers/FreeCAD-Ribbon #277: BUG: Ribbon bar text under each section in a workbench is obscured

SpeeQz1, 2025-02-04:
When using the Ribbon Ui 1.6.2.6 the text under each different section in a workbench is obscured when using different sizes for the icons such as the tab bar size:

[image]

[image]

APEbbers, 2025-02-04:
Hi,

Thanks, I noticed this last weekend. Can you test if it is fixed in either the develop branch or the pre-release branche? It might be solved in these

kind regards

SpeeQz1, 2025-02-04:
@APEbbers Develop fails to run the addon:

[image]

APEbbers, 2025-02-04:
That's strange. I can't found any module of PySide6. 

[image]

Perhaps you need another git pull? or change to Pre-Release 1.7.0 branche

SpeeQz1, 2025-02-04:
Pre-Release 1.7.0 does work and the text issues are fixed.

APEbbers, 2025-02-04:
Great!. I hope to release version 1.7.0 this week. I'm currently bug checking and have top update the manual with the menu changes
 What do you think of the menu changes? the goal is to integrate the ribbon more with FreeCAD

SpeeQz1, 2025-02-04:
I was busy for a while so I didn't get to touch the Ribbon UI much but it does seem nice. One thing I do miss is the circular icon for the FreeCAD menu, maybe add that as an option to switch from the normal one in the preferences?

[image]

Maybe also some sort of a way to modify the margin/padding of the icons in tab bar along with their size separately from the overall tab bar size and also the margin/padding for the other icons in the workbenches?

[image]

[image]

Another addition that would be handy is to edit the text font size of the sections:

[image]

I am still trying to figure out how to setup the dropdown buttons.

SpeeQz1, 2025-02-04:
Alright @APEbbers I figured out how to create a new panel with my custom dropdown button. Some things to note:

It is probably a good idea when a person removes a panel in the `Create new panels` menu the custom panel name to be replaced with an empty string or something like `Select a panel` so that a person doesn't accidentally delete a panel as the next panel in a row is chosen.

[image]

Also I managed to create duplicate identical custom dropdown buttons, probably would like to add a remove button and prevent that from happening.

[image]

SpeeQz1, 2025-02-04:
A small problem reported by another user is that minimizing the tab with the pin icon and then switching to a different workbench re-opens the workbenches instead of keeping them closed.

[image]

[image]

APEbbers, 2025-02-04:
Hi Thanks for the feedback! Please find below my anwsers

> Alright [@APEbbers](https://github.com/APEbbers) I figured out how to create a new panel with my custom dropdown button. Some things to note:
>
> It is probably a good idea when a person removes a [...]

Agreed. I'll add it with the next release.

> 
> Also I managed to create duplicate identical custom dropdown buttons which is possibly a visual glitch as they disappear once I try filtering them out, probably would like to add a remove button and prevent that from happening.

I'll look into that too.

> 
> Another problem I noticed with the Ribbon Layout settings is that the `Reset` button resets everything for the layout instead of resetting each individual page, also there is no `Cancel` button when clicking the `Reset` button and finally I believe if you click it accidentally you can't cancel the reset through the `X` button in the top right and it removes all your changes.

That is actually by design. The reset button replaces you json file with the default. to set it back in an initail state. I will add a question dialog with cancel button to it. By the way, i saw that you have the backups disabled. Perhaps a good idea to enable it. That way you you can restore your settings.

[image]

kind regards

APEbbers, 2025-02-04:
> A small problem reported by another user is that minimizing the tab with the pin icon and then switching to a different workbench re-opens the workbenches instead of keeping them closed.

This is by design actually. 
> 
> Additionally reported is some strange behaviour where hovering your mouse after clicking on another workbench causes some weird issues where the UI opens/closes itself.

Do you mean that the ribbon minimized itself when you move your mouse away from the ribbon? If so that is by design as well.

> 
> Additionally the `Help` menu is missing (don't know if any other is) from the FreeCAD menu button:

You are probally on the pre-release branch. For the next release, the help menu is moved:

[image]

[image]

The ribbon settings are here:

[image]

Ribbons overlay function:

[image]

kind regards

raindropsfromsky, 2025-02-05:
> I was busy for a while so I didn't get to touch the Ribbon UI much but it does seem nice. One thing I do miss is the circular icon for the FreeCAD menu, maybe add that as an option to switch from the normal one in the preferences?

Personally I prefer the current form. I remember people complained that the menu was "hidden" under the FC logo.
It is allright if this is an option, but the default form should be as it is now, IMHO.

> > A small problem reported by another user is that minimizing the tab with the pin icon and then switching to a different workbench re-opens the workbenches instead of keeping them closed.
> 
> This is by design actually.

This is smart! 👍 

> That is actually by design. The reset button replaces you json file with the default.

Can you reset only the changes on the current screen?
The user may have invested too much to dump everything in one shot! 

> Another addition that would be handy is to be able to edit the text font size of the sections:
> 
> [image]

I second this request! I have often felt that the text is too tiny or disproportionately sized.
Separate size controls should be provided for each level of the menu system also.

SpeeQz1, 2025-02-05:
@APEbbers what are thoughts about the rest of the of the buttons from the modeling panel being hidden in the panel options?
Is there any way preventing them from being hidden and to appear along with the rest of the buttons by default?

[image]

[image]

APEbbers, 2025-02-05:
Hi,

I had looked into the yesterday. i did create a fix for it. It's already in the develop and pre-release branch. I'm testing it further this evening. But perhaps you can give it a try already?

SpeeQz1, 2025-02-05:
@APEbbers checked `Pre-Release` and there were no changes and the rest of the buttons remained in the panel options button. Same for the `Develop` branch.

APEbbers, 2025-02-05:
Strange, this is what I have on Windows, linux and macOS:

[image]

Which sizes are you using?

SpeeQz1, 2025-02-05:
I was using different sizes but after resetting it the same problem occurs with the icons being put away in the dropdown menu.

SpeeQz1, 2025-02-05:
Oh, that fixes the issue thanks.
What do you think about setting `0` as the default value?

APEbbers, 2025-02-05:
> Oh, that fixes the issue thanks. What do you think about setting `0` as the default value?
Well people were complaining that the ribbon was to wide, etc. and didn't seem to use this function. So I set the default value to 6. For most workbenches, this shows most command. 

> Also another question, why do the buttons have smaller maximum values you can input before it prevents you from making it any bigger?
> 
> E.g. `Size of tab bar` can be only big up to 36.

I did that because of the padding and margin  issues. For that reason the under limit is 16 for all size settings.

closed 2025-02-07 by APEbbers

APEbbers, 2025-02-08:
Hi @SpeeQz1,

I've released version 1.7.0. For the moment settings to change the font size by the user are not in. I have however added it to my roadmap for version 1.8.0. ( [https://github.com/APEbbers/FreeCAD-Ribbon/wiki/07-%E2%80%90-Road-map](url) )

In version 1.7.0 I did make some improvements regarding the font sized. The fonts for the panel and buttons are set to pixelsize 11 and for the tabbar this is 14.  I've tested and these settings are consistent between Win11, Linux mint and macOS High Sierra

kind regards