APEbbers/FreeCAD-Ribbon #280: When I change the branch to 1.7, the Addon Manager reports uncommitted local changes

raindropsfromsky, 2025-02-05:
Once I change the branch to 1.7, the Addon Manager is unable to remember that.
So if I close FC, the branch reverts to "main" (1.6.2.6) again.

I have reported this bug in the FC Github.
(There are many other issues with the Addon Manager too.)

Anyhow, back to our issue:

When I change the branch to 1.7, the Addon Manager reports a warning like this:
[image]

The number of uncommitted changes vary each time.

I have no clue what this means.
- FC does not list these uncommitted changes (or provide a list so that I can quickly check them out).
- What does FC mean that I bring those changes "with me"??
- What is the downside of "_bringing the uncommitted changes with me_"?

I know the branching concept is meant for beta testers, but then the testers are doing a blackbox testing.
So they may not be aware of (or capable of-) doing a whitebox testing (by checking the change in the code).

s0, what is the bottomline?
Should I report this warning whenever I see it?
Or are those changes in the pipeline, so it does not matter?

APEbbers, 2025-02-05:
Hi,

You don't have to report it, because the problem is locally on your computer.
if you have git installed you can do a git revert --no-commit. This should fix it. You have to be on the develop branch though.

The cause is that locally on your computer a file is changed by something. Git sees that and that results in this error.  
[image]

otherwise click yes and see if the problem persists after update again

kind regards

raindropsfromsky, 2025-02-05:
oh!!

Actually I installed git only for the sake of using the development branch. I am not a coder and I do not use git myself.

closed 2025-02-05 by raindropsfromsky