APEbbers/FreeCAD-Ribbon #281: BUG: When I toggle overlay (F4) twice, the layout changes

raindropsfromsky, 2025-02-05:
My FC layput was set to show the panels as follows:
1. Tree panel - left top
2. data panel - left bottom
3. Tasks panel - right
4. Report panel - turned off (appears occasionally at bottom)

But I tried out the toggle overlay (F4). 
When I toggled it off, I am seeing a very strange layout, which I don't like at all.

[image]

I must confess that I am not clear what this command does.
Actually I tried that out to see what is does.

But at least it must not change the layout.
Even if it does, when I toggle the mode off, it should revert to my original layout.

-----------

How do I restore my favorite layout?

APEbbers, 2025-02-05:
it is by design, because it is basically the overlay function of FreeCAD with the ribbon basically excluded.
See my answer for #282.

kind regards

APEbbers, 2025-02-05:
Hi,

I've modified the overlay function. Now the position of the panels remain as they are. The layout won't change anymore

kind regards

raindropsfromsky, 2025-02-06:
I just checked out the Addon Manager, but there are no Ribbon updates.
Secondly, you have reopened the issue.
Does that mean that the resolution didn't work out?

APEbbers, 2025-02-06:
Hi,

Perhaps if do a git pull from commandline helps. I personally do my updates via the command line often. That way I see what is happening. I use the addon manager to see if there are updates for other workbenches.

I wanted to close this issue with the release. I just clicked to soon on close

closed 2025-02-07 by APEbbers

raindropsfromsky, 2025-02-08:
I have updated both Ribbon and FC.
The problem is still there.

Please reopen the issue.

raindropsfromsky, 2025-02-08:
I updated FC and Ribbon.
The problem is still there.

[image]

Please reopen the issue.

APEbbers, 2025-02-08:
Hi ,

I tested the function on Windows, Linux and macOS and it is working. Some time, the focus changes and you have to click twice but that seems an FreeCAD issue.
I have updated the roadmap for version 1.8.x. and I added improvment of the overlay function.

However, are you aware that you can toggle the overlay function per panel? This is a FreeCAD function:
[image]

kind regards

raindropsfromsky, 2025-02-09:
Please my remarks in issue #282