APEbbers/FreeCAD-Ribbon #282: Feature request: Show the selection state for the overlay and transparency menus

raindropsfromsky, 2025-02-05:
The menus do not show the current state of the overlay and transparency controls.
[image]

Even the labels of the menu options are ambigious (a "toggle" can be on or off.)
So putting a tick against the option does not tell us whether it is on or off.
(For example, what does it mean if I say "_toggle overlay- Yes_"?)

**Desired:**
1. Show a tick if the state is selected
2. Remove the word "toggle" from both menus, so that if a tick appears against it, there is no ambiguity.

APEbbers, 2025-02-05:
Hi, this is basically the overlay function of FreeCAD without the ribbon. I just changing some parameters under tools->edit parameters. Therefore a toggle state is not possible.

You can see if the overlay is toggled, because the panels will be transparant. (like the tree). When the panels are transparant, you can toggle transparency  to enable or disable transparency when hovering over them.

https://github.com/user-attachments/assets/6f1a061f-b742-4036-b7c4-7fcaa8320283

Currently, the transparency is not toggling, but only disables transparency. I'll look into that.

kind regards

closed 2025-02-07 by APEbbers

raindropsfromsky, 2025-02-09:
I found a very brief description [here](https://wiki.freecad.org/Std_View_Menu).
It is just not sufficient to explain how this function works.
Besides, the description does not state that transparency control works only if the overlay mode is turned on.

I could not find any YouTube video on this subject, either. 

Could you please elaborate how the two controls work individually, and when used together?
That will allow me to test this thoroughly.

BTW, the manual page describes that there are many Overlay controls. (left, right, etc.)
But in Ribbon, there is only one overlay control. How come?

APEbbers, 2025-02-09:
> Could you please elaborate how the two controls work individually, and when used together?
That will allow me to test this thoroughly.

They only work together. The manual is actually saying it: 
[image]
If there is no overlay panel, the panel won't be transparant on all times.

> BTW, the manual page describes that there are many Overlay controls. (left, right, etc.)
But in Ribbon, there is only one overlay control. How come?

Not in yet. Is not that important but is on the roadmap. all panels have these buttons anyway:
[image]

The function is quite new. Only present in FC 1.0. That's why there are not that many videos of it. It is a feature from realhunter.

kind regards,