APEbbers/FreeCAD-Ribbon #283: BUG: The transparency menu does not work on any of the panels (left side, top, right side)

raindropsfromsky, 2025-02-05:
The transparency menu does not work.

[image]

In the following experiment, I placed three objects partly under the three panels (left, top, right).
Then I toggled the transparency.

[image]

I expected all the panels to become semitransparent, so that we can view the objects under them.
But this does not happen.

I know that this control is not supposed to toggle the transparency of the objects themselves.
(For that function, its placement would be in the menus, not in the Windows bar.)

APEbbers, 2025-02-05:
Hi, see my answer for issue #282.

Overlay has to be enabled first.

closed 2025-02-05 by APEbbers

raindropsfromsky, 2025-02-05:
> Overlay has to be enabled first.

If transparency can be set only when overlay is set, then the transparency option should be presented as a suboption of overlay.

APEbbers, 2025-02-05:
This is FreeCAD menu structure for overlay. I'm following that at the moment:

[image]