APEbbers/FreeCAD-Ribbon #293: Feature request: Reveal on hover feature for the ribbon.

raindropsfromsky, 2025-02-08:
When the Ribbon is hidden, the user gets a much larger working area. So it is desirable to autohide the ribbon.

However, to reveal the ribbon, the user has to click on the WB toolbar button for the same WB.
If he clicks on any other WB, that WB is shown.

This is inefficient workflow, because the user has to specifically seek and click on the current WB button.

Instead, let the user use a mouse gesture to move the mouse toward the WB toolbar.
As soon as the mouse touches anywhere on the WB toolbar, reveal the ribbon.
(Here, he should not be required to hover on the specific button of that WB. Hovering anywhere on the toolbar should be enough.)

If the user moves the mouse below the ribbon area, FC should autohide the ribbon again (just like now).

APEbbers, 2025-02-08:
Hi,

Well in the past there were issues with this function and the overlay functions. I've enabeled it again on the develop branch.
Are you willing to test it for a while? If it works, I'll make it an option in the next release. 

In the past the ribbon disappeared and showed erratic behaviour.  That's why it is disabled for FC1.0

Kind regards

raindropsfromsky, 2025-02-08:
> Are you willing to test it for a while? If it works, I'll make it an option in the next release.

Of course! Let's do it!

If you can add some debug messages, I can send you the logs.

APEbbers, 2025-02-08:
There won't be any crash or anything. Its just that the ribbon shows erratic behaviour and completely disappears.

See [https://github.com/FreeCAD/FreeCAD/issues/17500#issuecomment-2439623348](url) for a short screen capture of the problem.
If there is a crash, it will be registered in the logs.

APEbbers, 2025-02-08:
Hi,

I've enabled the setting for enabling this behaviour. Incase it is not working check if setting below is enabled:
[image]

raindropsfromsky, 2025-02-09:
> See [https://github.com/FreeCAD/FreeCAD/issues/17500#issuecomment-2439623348](url) for a short screen capture of the problem.

Github says "no content".

> There won't be any crash or anything. Its just that the ribbon shows erratic behaviour and completely disappears.
Sure, I will check this out.
- How frequent is the disappearance?
- How do we get it back? (Or do we have to restart FC to get it back?)

I asked DeepSeek, and got this response:

The logs should capture:
- Ribbon initialization and destruction.
- Visibility changes.
- Parent window or container changes.
- Error conditions.
- UI thread and synchronization issues.
- User actions and system events.
By adding the following logs, the developer can help you trace the root cause of the ribbon disappearance. 

It provided this sample code:

```
private void LogRibbonState()
{
    string visibility = ribbon.Visible ? "Visible" : "Hidden";
    string parentState = ribbon.Parent != null ? "Parent attached" : "Parent detached";
    string logMessage = $"Ribbon State: {visibility}, {parentState}, Thread ID: {Thread.CurrentThread.ManagedThreadId}";
    Logger.Log(logMessage);
}

private void Ribbon_VisibilityChanged(object sender, EventArgs e)
{
    string visibility = ribbon.Visible ? "Visible" : "Hidden";
    string logMessage = $"Ribbon visibility changed to: {visibility}. Triggered by: {GetTriggerSource()}";
    Logger.Log(logMessage);
}

private void CheckRibbonPeriodically()
{
    if (!ribbon.Visible)
    {
        Logger.Log($"Ribbon unexpectedly hidden. Last known state: {ribbon.Tag}, Last user action: {GetLastUserAction()}");
    }
}

// Call this method periodically using a timer or background thread
private void StartRibbonMonitoring()
{
    System.Timers.Timer timer = new System.Timers.Timer(5000); // Check every 5 seconds
    timer.Elapsed += (sender, e) => CheckRibbonPeriodically();
    timer.Start();
}
```

raindropsfromsky, 2025-02-09:
> I've enabled the setting for enabling this behaviour. Incase it is not working check if setting below is enabled: [image]

I checked for an update in AM. 
But it did not show a new update.

So I uninstalled Ribbon WB in order to re-install it.
But here, I faced a problem: FC reported this:

```
Unknown error:
Some files could not be removed.

```

After this, I restarted FC.

It showed a strange UI: All the toolbars had vanished!
There was only the menu, and nothing below it.

I re-installed Ribbon.
This time it has installed 1.7.1.

But the configuration of the ribbon has changed a lot.
(It did not remember the old settings.)

[image]
- The quick commands have re-appeared.
- The ribbon now shows all WBs, not the limited number I had selected. 
   The toolbar is much longer now, with even overflow icons (<>) showing.
- The sections now prefer "width-first" arrangement of icons.
   Earlier, all icons were fitted in "column-first" mode. 
   (Arrange all icons in the first column. If there are more than 3 icons, they are placed in the second column, and so on...)

When I selected the **Ribbon Layout** menu, it triggered a "first-time use" dialog!
[image]

When I chose "NO", it produced this error:

```
---------------------------
Critical error occurred
---------------------------
Oops. An unexpected error occurred:
```
  File "C:\Users\naray\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\FCBinding.py", line 1332, in loadDesignMenu
    LoadDesign_Ribbon.main()
  File "C:\Users\naray\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\LoadDesign_Ribbon.py", line 5119, in main
    Dialog = LoadDialog().form
             ^^^^^^^^^^^^
  File "C:\Users\naray\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\LoadDesign_Ribbon.py", line 193, in __init__
    with open(DataFile, "r") as file:
         ^^^^^^^^^^^^^^^^^^^

FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\naray\\AppData\\Roaming\\FreeCAD\\Mod\\FreeCAD-Ribbon\\.\\RibbonDataFile.dat'
```
---------------------------
OK   
---------------------------

```

raindropsfromsky, 2025-02-09:
First-cut reaction: WOW! 👏 

I experimentally changed the WBs a few times, and it works fantastically well.

However, should the side panels also not have the same behavior?
(Or perhaps I am thinking of seeing it in Realthunder's fork.)
Anyhow, can it be added?

raindropsfromsky, 2025-02-09:
In some WBs, the arrangement of buttons looks odd. 

[image]
A better placement algorithm is needed to beautify it.

APEbbers, 2025-02-09:
Hi,

Thanks for testing. Great to hear that you like it! Just keep using it for a while and if anything happens, let me know!.

> How frequent is the disappearance?
How do we get it back? (Or do we have to restart FC to get it back?)

if the ribbon suddenly disappears, just notify me. The root of the problems are with FC's overlay function. I'm hoping now the ribbon has its modified version that exclude the ribbon, that the issues are gone.

> I asked DeepSeek, and got this response:

The logs should capture:

Ribbon initialization and destruction.
Visibility changes.
Parent window or container changes.
Error conditions.
UI thread and synchronization issues.
User actions and system events.
By adding these logs, the developer can help you trace the root cause of the ribbon disappearance.

I know how to write logs thats no problem. It is in on several places in the ribbon already actually. But it is an FreeCAD issue. Just the standard FC logs are just fine for now.  

> When I chose "NO", it produced this error:

Regarding the error message on the layout menu: good catch! I'll fix that.

> I re-installed Ribbon.
> This time it has installed 1.7.1.
> 
> But the configuration of the ribbon has changed a lot.
> (It did not remember the old settings.) -----> Is this intended?

Regarding the re-install of the ribbon. You get the initial layout that is shipped with the Ribbon. That includes the quick access toolbar. You basically did a hard reset.
Keep in mind that large button has one row, medium buttons two rows and small buttons three rows. Looking at your picture, the sizes of the medium and large buttons equal. Sections 3 are large buttons, sections 4 and 5 have both.

Perhaps instead of re-installing the ribbon, use git from the command line instead. When the add-on manager is not updating, this will work. Personally, I use Tortoisegit for this. With Tortoisegit you can do the git pull from windows explorer.
[image]

> However, should the side panels also not have the same autohide feature?
Or perhaps I am thinking of seeing it in [Realthunder's fork](https://github.com/realthunder).
Anyhow, can it be added?  

That is already in FreeCAD! If overlay is enabeled you can select autohide:
[image]  

> In some WBs, the arrangement of buttons looks odd.
A better placement algorithm is needed to beautify it.  

This is because of the very long names for the panels. Not sure why the developer did this. But you can change the titles yourself in the layout manager:
[image]

kind regards

raindropsfromsky, 2025-02-09:
> Keep in mind that large button has one row, medium buttons two rows and small buttons three rows. Looking at your picture, the sizes of the medium and large buttons equal. Sections 3 are large buttons, sections 4 and 5 have both.

That explains it!
After seeing the "first-time use" window, I should have realized that the usual mix of button sizes will reappear.
I missed that cue!

I actually expected FC to reflow the buttons when all of them are made small, so that all them are packed in stacks of 3.

BTW why should we have large and medium buttons at all?
All buttons have the same priority and frequency of use, after all!
(I can appreciate that frequently used buttons are made larger so that they present an easier clicking target.)

Yes, I have used the same size for large+medium+small buttons. (fantastic observation there!)

> I use Tortoisegit for this. With Tortoisegit you can do the git pull from windows explorer.
Thanks for the tip. I will use a Git frontend, as I am not comfortable with commandline, or explorer extensions.

> But you can change the titles yourself in the layout manager:
Thanks for this tip!
Yes, I will certainly do this. It's an eyesore!

> [The autohide] is already in FreeCAD! If overlay is enabeled you can select autohide:
Thanks for the tip!

BTW that reminds me: The selection state of the **Overlay** and **Transparency** menus is still not visible.
[image]
Please expedite this. Thanks in advance!

raindropsfromsky, 2025-02-09:
If I want to make all icons small, it is a laborious task:

I have to repeat the following step for all sections of all workbenches:
[image]

The alternative is to use script, which forces me to reach under the hood.
Non-coders like me will hate that approach.

Instead, why not provide an easy way?

Just add an "all" entry in the WB drop-down manu, and also in the sections drop-down menu.
[image]
That way, I can select all sections in all WBs, and set small icons for all of them in one shot!

> This is what I have in mind: a proper CAD interface like SolidEdge
Yes, I had already guessed it. But the same questions apply to them as well!

APEbbers, 2025-02-09:
Why don't use the initial setup tab? there you can set everting to small:
[image]

raindropsfromsky, 2025-02-09:
> That is already in FreeCAD! If overlay is enabeled you can select autohide:
It does not work for me!

If I press F4, all the panels vanish on left and right.
When I press F4 again, a different set of panels appears.
Subsequent presses of F4 toggles these new panels on/off.
My original layout is lost forever.
I have to drag and drop the panels all over again.

This is a major nuisance!

When I toggle F4, these are the headers of the panels as I see them:
[image]
[image]

 But I have never seen the icon you have shown.

APEbbers, 2025-02-09:
> > That is already in FreeCAD! If overlay is enabeled you can select autohide:
> > It does not work for me!
> 
> If I press F4, all the panels vanish on left and right. When I press F4 again, a different set of panels appears. Subsequent presses of F4 toggles these new panels on/off. My original layout is lost forever. I have to drag and drop the panels all over again.
> 
> This is a major nuisance!
> 
> When I toggle F4, these are the headers of the panels as I see them: [image] [image]
> 
> (The headers have not changed at all!) But I have never seen the icon you have shown.

I don't know how else to explain the overlay function. When you toggle overlay, the panels are set to autohide.
This is very much a feature of FreeCAD itself. the Ribbon version only excludes the ribbon. For all other panels, it uses FreeCAD functionality. 
You might look into FreeCAD's preferences and set them like this:
[image]

also try to toggle this:
[image]

raindropsfromsky, 2025-02-09:
> Why don't use the initial setup tab?
oops!! Why didn't I think of that? 🤦 

It works fine now.

-----

I found the missing panels!
After I press, all panels vanish.
But they pop up in the form of a tab bar on the left/right edge of the monitor.
Here is the screen capture for the left side tab bar.
[image]

The hot zone for hovering extends just 1" below the ribbon.
If we push the mouse pointer below that, the tab bar won't pop up.
As a result, the user has to move the mouse by a large distance.
This design also takes a mouse hover and two extra clicks. 

In comparison, RealThunder's design allowed us to take the mouse over any part of the panel, and show the panel without a single click.

APEbbers, 2025-02-09:
> The hot zone for hovering extends just 1" below the ribbon.
> If we push the mouse pointer below that, the tab bar won't pop up.
> As a result, the user has to move the mouse by a large distance.
> This [...]

I think that is a difference between  RealThunder's design and FreeCAD 1.0. It might have to do with the FreeCAD themes.
It is not related to the Ribbon. I'm only excluding the ribbon from the FreeCAD Overlay.

You can set the hint to show the tabs permanent. (Works partly)
[image]

raindropsfromsky, 2025-02-09:
I discovered a new problem: While tinkering with panels, I turned off Ribbon. Now the menu itself is gone!
I restarted FC, but the window is a plain black frame!

How to restore the Ribbon??

APEbbers, 2025-02-09:
if you delete these files, your FreeCAD settings will be reset to default. Including UI settings:

[image]

raindropsfromsky, 2025-02-09:
oof! Right back to square one! 

This should not happen!
When Ribbon is turned off, the native menu system of FC should be restored.

APEbbers, 2025-02-09:
Well, that is not yet possible. However, you can also try to toggle overlay again. if you right click on normal panels, you should be able to select the ribbon again. (just tought of it)

Have created a copy of these files first? perhaps if then restore user.cfg not all is lost.

raindropsfromsky, 2025-02-10:
> But you can change the titles yourself in the layout manager:
> [image]

I tried to do this, but could not find the right place to carry out the label editing.
These are the labels I'd like to change.
[image]
Which part of the Layout Editor allows me to do it?

APEbbers, 2025-02-10:
Hi 

sorry, I posted the wrong solution. Use the combine panels tab instead, and select one panel only. You then set your own name. And the panel will be replaced with your "new" panel.

kind regards

raindropsfromsky, 2025-02-10:
I followed this, and created three "new" panels.
But I am unable to see them anywhere.
They are not within the Assembly exploded WB.

I also faced this error. 
[image]

FC went into a lock, and would not respond at all.
I had to kill it.

raindropsfromsky, 2025-02-10:
In fact, what is the purpose of the **Add** button? Is it not redundant?
IMHO after the 4 steps shown above, if I click **Update**, it should complete the renaming of the panel.
(Effectively I created a clone of the panel with a new name.)

After this, the UI does not allow me to repeat the same 4-step sequence for the next panel.
How do I take up the next panel? Do I have to close the whole Layout dialog and restart FC before doing it from the beginning?

raindropsfromsky, 2025-02-10:
I used the **Combine panels** workflow to rename the **ExplodedAssemblyAnimationControlTools** panel.
When I restarted FC, it shows the newly renamed panel, but it is blank!

[image]

raindropsfromsky, 2025-02-10:
I checked the **Ribbon Design** section.

[image]
The newly renamed panel appears there, but there are no buttons in the table.

APEbbers, 2025-02-10:
Hi,

Did you create a datafile the first time you opened the layout menu?

I'll look in the procedure tonight.

kind regards

raindropsfromsky, 2025-02-10:
> Did you create a datafile the first time you opened the layout menu?

Well, if FC creates that file, I don;t know.
But the UI does not lead me through the steps needed to create and save a datafile.
I was not even aware that the user has to create a datafile.
(BTW what is it? What format? What does it contain?)

APEbbers, 2025-02-10:
Hi,

Please read the manual on how to design the ribbon.
[image]

kind regards

raindropsfromsky, 2025-02-10:
I went through the [documentation about combining panels](https://github.com/APEbbers/FreeCAD-Ribbon/wiki/03%E2%80%90-Ribbon-design#combine-panels).

The process is intuitive, and what I did was exactly right.

Except for the last step: 
The procedure does not describe how to create another combined panel without quitting the **Layout** window.

Also, it is OK to refresh the datafile in the beginning. The refresh button is provided at the very top, so this comes naturally to the user. But after making one change using any one tab (e.g. combine panels, exclude panels), is the user supposed to refresh the datafile before attempting any iteration of the same tab, or another tab?

Logically, I would click on the Update button at the bottom, and return to the next iteration in the same tab.

For example, after combining some panels, click on the Update button, and then return to the top of the screen to combine some more panels. However, at this time, the screen already has the data loaded from the previous iteration. How do I clean it up? 
- There is no "move left" button on the divider, to clear up the right pane.
- There is no "**Clear**" button to clear up the right pane.
- The only logical way seems to click on the **Close** button to exit the dialog.
   But that method seems to be too convoluted! Would there be a quicker method?

To sum up, the process is very intuitive till the end.
But it does not provide an intuitive workflow if the user wants to save his changes and continue making more changes.

The documentation also does not address this workflow.

APEbbers, 2025-02-10:
Just set the top dropdown on new:
[image]

That clears the right pane:
[image]

raindropsfromsky, 2025-02-10:
> what is the purpose of the **Add** button? Is it not redundant?

Even after reading the manual, I still have the same thought: Why is it necessary to have this button at all?

Imagine this workflow:
1. The user enters a name for the new (combined) panel.
2. He selects the panels to be combined, and presses the right arrow.
3. Then he saves this definition in one stroke. 

(I have already pointed at this ambiguity in the UI: It does not have an obvious "save" button.)

raindropsfromsky, 2025-02-10:
> The add button saves the panel internally and added to the dropdown. You can then create another panel with a different name.

That is precisely the problem!
The definition of a new Combined panel has two parts:
1. a name and 
2. constituent panels.

So, logically, the **Add** button should be below (or to the right of-) the right pane, which shows the constituent panels.

But here, the **Add** button is placed against the name only.
So it is associated only with the Name field, not with the "constituent panels" field.
When the user clicks on the **Add** button, he has not even defined the constituent panels.
So he will think that only the name is saved; not the constituent panels!

**Desired workflow/layout:**
1. Move the Add button to the right of the constituent panels field (or below it)
2. When the **Add** button is clicked, two things should happen: 
    (a) The newly defined panel should get saved, and 
     (b) all fields should be cleared off in the form.

raindropsfromsky, 2025-02-11:
Hi

the "reveal on hover" issue is already resolved. 
But we have started disucssing all the issues that are discovered subsequently in this thread.
Now multiple issues were getting jumbnled up in one place.
So I have started reporting each issue separately.

I hope this is ok.

closed 2025-02-11 by APEbbers

raindropsfromsky, 2025-02-11:
I installed the latest weekly, and tested this once again.
The problem is still there.

[image]

Even after I close the dialog, the "restart to let the change take effect" dialog is missing.
FC does not restart on its own.
So I am stuck with an old version of Ribbon (1.7.1) now!

**Update:** 
I tried this again, and FC responded this time.

But now the ribbon has vanished!
[image]

I tried this once again, and restarted FC once more.
It has the same problem.

I know that I have to delete files in the appdata> FC path, but why is this happening all of a sudden?