APEbbers/FreeCAD-Ribbon #294: "Critical Error" on switching to some workbenches' tabs

morganwk, 2025-02-10:
I'm running FreeCAD on Kubuntu, installed from a flatpak, most recent version. I have a relatively clean install with only a few other addons installed, and the addons in question don't appear to have any issues.

Whenever switching to certain workbenches with this addon active, I get a dialog box reporting a "critical error" which comes back up when dismissed, effectively soft-locking FreeCAD entirely until I restart it. Currently, I've determined that Assembly and FEM trigger this every time I've tried loading them, but a few others don't. Here's the error:

```
Oops. An unexpected error occurred:
  File "/home/mogan/.var/app/org.freecad.FreeCAD/data/FreeCAD/Mod/FreeCAD-Ribbon/./Resources/packages/pyqtribbon_local/tabbar.py", line 97, in changeColor
    currentTabColor = self._tabColors[currentTabText]
                      ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^

KeyError: '&Part Design'
```

So far I've only observed this on one computer, but I'm going to be at my local makerspace tomorrow so I'll report back with further info if the problem occurs on Windows as well. Otherwise I'm unsure of how to proceed.

APEbbers, 2025-02-11:
Hi @morganwk ,

I couldn't reproduce the error on my windows laptop. Had succes testing it on windows?

kind regards

morganwk, 2025-02-12:
Looks like it might be a flatpak issue, or I might have had leftover corrupted config files / config files from old versions of either freecad or the plugin somewhere. I just tried again using the appimage version and it's working fine. It also works on at least one Windows install I tried.

I'll reply again or open another issue with more detailed information if it happens again.

APEbbers, 2025-02-16:
Hi @morganwk,

I've not been able to reproduce this with the develop branch in Linux Mint with FlatPack.
Can you test if the issue is resolved with the develop branch?

kind regards

APEbbers, 2025-02-16:
No problem! Great to here that it is working again.

Kind Regards

closed 2025-02-16 by APEbbers